raspberrypi/rpi-imager #209: Provide a simple way to delete cached image

ralphrmartin, 2021-04-13:
If you only want to use this to set up one Raspberry Pi, it makes no sense to keep the cached download, which can waste a significant amount of disk space. It is not obvious to users that the file is being cached - I assumed it would be deleted on exit (or perhaps be in /tmp). Please provide some simple user interface for deleting it, e.g. a checkbox "Remove download after installing".

maxnet, 2021-04-13:
But if you only ever want to use it to setup that single Pi, does it make sense to keep Raspberry Pi Imager installed?
The uninstaller does remove it...

>I assumed it would be deleted on exit (or perhaps be in /tmp)

Technically the program does not need any temporary files to function. Can stream files from the Internet straight to SD card, uncompressing on the fly.
Caching was added later.

Problem is the UI needs to beginner friendly, leaving out configurable options.

ralphrmartin, 2021-04-13:
The standard, Apple way, to uninstall stuff on a Mac is to drag the application to the trash, so that won't delete the cache. It does not have uninstallrs like Windows and Linux.

I hardly think it's going to confuse even beginners to say "Keep a copy of the downloaded Raspbian OS for further installations in future".

I'd say its even less friendly to silently waste a few GB, especially for users with small disks.

maxnet, 2021-04-13:
>The standard, Apple way, to uninstall stuff on a Mac is to drag the application to the trash, so that won't delete the cache.

Yeah, that do is a hard to solve problem.

Apple only cleans up properly for apps that come from Mac Store.
But our application is not welcome there (they do not allow any applications that needs to do anything privileged there...)

alfieroddan, 2021-08-30:
I can't actually find the cached image on my Mac. 

Is there a standard location for the Rasperry Pi Imager to store images? 

I agree with @ralphrmartin that even a simple delete cache on the UI would be helpful. I am said "beginner" on booting different images on to my pi. 

Love the app so far tho.

lurch, 2021-08-31:
I think RPi Imager only caches the last-flashed image, so it doesn't matter if you flash multiple different images onto your SD card. (so I guess a sneaky way to reduce the amount of disk-space used by the cache would be to flash one of the smaller OSes? :wink: )

maxnet, 2021-09-03:
>I can't actually find the cached image on my Mac.

Menu "Go" -> "Go to Folder..." -> ~/Library/Caches/Raspberry Pi

raisaab198, 2021-10-03:
This is really so annoying that I am using just 25GB of hdd for running my Manjaro Linux as dual boot and this cached image has occupied 2.8gigs of storage and I didn't found this cached image anywhwere to delete it. If somone found the way, please share. Thank You

raisaab198, 2021-10-03:
In Linux, you can simply select show hidden files and then open /home/user/.cache/raspberry pi and then simply delete the lastdownload zip file.

beckmx, 2022-01-06:
hello, I would like to erase the cache image, but I cannot find it under the "Caches" on macos

maxnet, 2022-01-06:
>hello, I would like to erase the cache image, but I cannot find it under the "Caches" on macos

Menu "Go" -> "Go to Folder..." -> ~/Library/Caches/Raspberry Pi

At least the "Raspberry Pi" folder should always exists there.
If you cannot find a lastdownload.cache file under there, it do can be that there is no cached file. 
It only caches if there is at least 5 GB + size of image space available, and you succeeded in writing the image at least once.

beckmx, 2022-01-06:
@maxnet thanx, well I have 3gigs left but I used to have more before I flash the image, and the "rpi imager" says, "Cached on your computer" but no folder(not hidden) was there. I'll take ur answer as valid then

maxnet, 2022-01-06:
Double check from terminal, just in case "Finder" is configured not to show certain files.

```
Maxs-Mac-mini:~ max$ ls -l ~/Library/Caches/Raspberry\ Pi
total 0
drwxr-xr-x  5 max  staff  160 21 dec 00:22 Imager
Maxs-Mac-mini:~ max$ ls -l ~/Library/Caches/Raspberry\ Pi/Imager
total 2350344
-rw-r--r--    1 max  staff  1203372157 21 dec 00:23 lastdownload.cache
drwxr-xr-x    4 max  staff         128 27 feb  2020 oslistcache
drwxr-xr-x  248 max  staff        7936 22 dec 19:40 qmlcache
```

Should also be able to elete from shell: 

```
rm ~/Library/Caches/Raspberry\ Pi/Imager/lastdownload.cache
```

varac, 2022-05-31:
Where does the imager store it's cached images on Linux ?

raisaab198, 2022-05-31:
It's in your home folder in .cache/rpi-imager

g4r37h, 2022-06-10:
> In Linux, you can simply select show hidden files and then open /home/user/.cache/raspberry pi and then simply delete the lastdownload zip file.

Yeah that's great. You can do that in all OSs. But it would clearly be better for the the util provide a way to either

a) not cache; or
b) delete the cached image itself

lurch, 2022-06-11:
Mildly amusing that this issue is asking for "less caching" and #300 is asking for "more caching" :wink:

g4r37h, 2022-06-11:
They are not mutually exclusive issues. Both point to the fact that the utility is lacking in functionality.