raspberrypi/rpi-imager #211: Feature Request

ComputerDoktor, 2021-04-16:
add automatic decompression for bzip2 .bz2 and zstd .zstd

maxnet, 2021-04-16:
Not sure if the interest in those are worth the extra trouble on Windows and Mac OS X to add the extra dependencies.

On Linux every archive format supported by libarchive should already work out-of-the-box.

ComputerDoktor, 2021-04-16:
Just an idea to help with the goal  "make Imager as easy to use as possible". Personally I know how to use bzip2 and zstd under Windows, although the decompression adds another timeconsuming step. But for a newbie bzip2 decompression may represent a hurdle.

lurch, 2021-04-16:
How commonly are OS images distributed in bzip2 or zstd? :thinking:

maxnet, 2021-04-16:
>zstd might become more common because of decompression speed.

Note that according to [zstd's own tests](https://facebook.github.io/zstd/) .gz (zlib) does 400 MB/sec as well on modern hardware.
So decompression speed is not really a limiting factor when writing to your average SD card that may do 40 MB/sec.

mdevaev, 2021-10-20:
Zstd has a very high compression level and I would prefer to see it in the imager.

closed 2021-12-09 by maxnet

hecko, 2023-11-07:
> Not sure if the interest in those are worth the extra trouble on Windows and Mac OS X to add the extra dependencies.
> 
> On Linux every archive format supported by libarchive should already work out-of-the-box.

Yocto Linux is by default built into bz2

commit 67618a2e (2021-12-09): Bump bundled libarchive version to 3.5.2

- Update bunlded libarchive version used on Windows/Mac
- Enable requested zstd support while we are at it. Closes #211