raspberrypi/rpi-imager #212: Error reading from storage. SD card may be broken

dubnemo, 2021-04-18:
From the Forum:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=308286&p=1853427

I have two **new** Sandisc 32G microSD card and **new** Merkury USB Universal 33-in-1 reader.

On **both** Raspberry Pi OS and Windows 10 OS I and other users have received this error from the imager while running as sudo or Win admin mode:

Error reading from storage.
SD card may be broken.

Anti-virus is OFF on Windows, as I originally suspected a background process stopping the verify step.

I have two RPi3Bs that go into infinite rainbow screen loop with microSD cards based on their boot stripe state when receiving this error. 

Another user claims the may have solved with specific USB microSD reader on his Mac.

maxnet, 2021-04-18:
>I have two RPi3Bs that go into infinite rainbow screen loop

That's strange.
Imager will remove the partition table of the card at the start of provisioning, and will not put one back until verifying completed.
If it errors out with an "Error reading from storage" error, it should never reach the stage that the partition table is written, and the card should not boot at all.

Should not even get to rainbow screen on Pi 3 in that case, but get a blank screen.
If it does get to rainbow, then it could be that Imager did not succeed at removing partition table or perhaps even writing anything at all to the card, and it still has previous content.

lurch, 2021-04-18:
> Merkury USB Universal 33-in-1 reader

The reviews at https://www.target.com/p/targus-33-in-1-card-reader-writer/-/A-13567347 suggest it's not a very reliable card-reader? :shrug:

dubnemo, 2021-04-19:
@lurch 

And you have recommendations for a reliable card reader/writer?  
This is what Microcenter had in stock.
I don't have infinite budget or infinite time to find the best one, especially if there a compatibility issues. 

I did not the the Windows Explorer opens up with the error occurs:
[image]

lurch, 2021-04-19:
I have neither infinite budget or time either, but I tend to buy known-brand stuff that looks decent, not whatever I can find in the bottom of the bargain bin :laughing: 
To be more serious, I think these "XX-in-1" devices tend to be fairly low quality, as they're trying to cram too much functionality into a low-cost device. I think dedicated SD-only card-readers might be more reliable? :shrug:

dubnemo, 2021-04-19:
@lurch 
Thanks for the reply. I see this post:
https://www.raspberrypi.org/forums/viewtopic.php?t=185154
And there is mention that 'speaking devices should be marked SDHC or SDXC compatible'.

Do you know much about this statement?

maxnet, 2021-04-19:
If in doubt what to get, I would at least get an USB 3 card reader, for no other reason that you have a higher chance it has a more recent chipset.

lurch, 2021-04-19:
You mentioned you're using a 32 GB card, which means it'll be SDHC. SDXC is only for cards 64GB and above. More info at [wikipedia](https://en.wikipedia.org/wiki/SD_card)

dubnemo, 2021-04-19:
@lurch @maxnet  I just purchased off Amazon the SanDisk - SDDR-B531-GN6NN USB 3.0 microSD reader/ writer.  
https://www.amazon.com/gp/product/B07G5JV2B5
If this works, I will close the issue.   Perhaps a good Wiki page for people to add their recommended/ proven readers would be helpful and REALLY close the issue. ;-)

lurch, 2021-04-19:
> Perhaps a good Wiki page for people to add their recommended/ proven readers would be helpful and REALLY close the issue. ;-)

Similar sorts of lists have been suggested before for other things, and the trouble is that there's so many different varieties that any such list would always be woefully incomplete / inaccurate. And there's so much "clone" hardware at the cheaper end of the scale, what works for one person may not work for another? :man_shrugging: 

The new reader you've bought looks like a good one - I've got one of those that came bundled with a SanDisk microSD card a while ago. The only tricky part is using your fingernail to get the card out again afterwards :wink:

dubnemo, 2021-04-20:
Maybe keep this open, and people can pile on with ones that DON'T work?

dubnemo, 2021-04-21:
The new SandisK USB 3.0 did provide a success write and verify on one 32G and one 16G microSD card.  However BOTH RPi3Bs go into infinite Rainbow screen loop -- I don't think both of these are defective.

closed 2021-04-22 by dubnemo

lurch, 2021-04-22:
Have these SD cards and Raspberry Pis been "known working" before, or are they all brand new? What power supply are you using?
If RPiImager was able to successfully write and verify the SD cards using the new card-reader, I guess that points to your old card-reader definitely being defective (hopefully you can get a refund) ? And presumably the problem isn't with RPiImager but with something else? Maybe something in https://www.raspberrypi.org/forums/viewforum.php?f=28 will help?

lurch, 2021-04-22:
(LOL, I typed my reply just before you closed the issue)

dubnemo, 2021-04-22:
@lurch classic low power situation...my bad.  Thanks a lot!

stoomey-coalnetwork, 2021-07-20:
Has someone found a workaround or fix for SD Card imager or Belana not working on a Mac OS Big Sur 11.4? All the products I try fail at the Verify stage. Thanks

lurch, 2021-07-20:
@stoomey-coalnetwork Are you using an M1 chip? Have you seen #235 ?

maxnet, 2021-07-20:
Do not think failing at VERIFY STAGE is related to the issue of Imager not starting on a M1 if an offer to install Rosetta has been previously denied.

Issues at verify stage are more common cuased by:

- broken SD card
- OR broken SD card USB reader
- OR broken micro SD card <-> full size SD card adapter thingy.

stoomey-coalnetwork, 2021-07-20:
> @stoomey-coalnetwork Are you using an M1 chip? Have you seen #235 ?

No. This is an Intel MacBook Pro

stoomey-coalnetwork, 2021-07-20:
> Do not think failing at VERIFY STAGE is related to the issue of Imager not starting on a M1 if an offer to install Rosetta has been previously denied.
> 
> Issues at verify stage are more commonly caused by:
> 
> * broken SD card
> * OR broken SD card USB reader
> * OR broken micro SD card <-> full size SD card adapter thingy.

I get your logic but this is also what I have found...

RPI Imager - on an SD card - gets to the end of the verify stage and then tells you the checksum doesn't match on SD Cards 
Balener Etcher - on an SD card - works as long as you don't let it finish the verify stage. If you let it verify the checksum doesn't agree and the disk isn't useable 
RPI Imager on an SD stick works 

SO I thought I might have an issue with the sd card but that same thing happens on 2 SD cards.... so then I thought it might be the SD slot .. but I bought a second raspberry pi and got the same result .... 

Now I have 2 SD Sticks, 2 Rasp Pi's, 2 SD Cards, 3 SD card readers and 3 apps on the Macbook to format a disk and still not got the sd cards working...... See my challenge here?

lurch, 2021-07-20:
Maybe there's some background software (security or anti-virus or similar?) running on your Mac, that is modifying the SD-card between the "flashing" stage and the "verifying" stage? :shrug:

maxnet, 2021-07-21:
It is normal that SD card is not usable if verification fails, as in that case we will not write out partition table by design.

Don't think background software is the problem, as that would be likely to give same problem when writing usb stick, that does work.

dubnemo, 2021-07-21:
SDHC reader/writer was the solution. The reader I had was not compatible, the Sandisk 3.0 writer was the fix.

shreyashkashyap, 2022-03-07:
HELLO GUYS 

first you need to check the SD card , if it is physically OK then you need to change the SD Card riders 

I am facing the same issue ,I try all the solution but my issue was not fix , even i change the SD card , at the end I change the SD card Reader and with in an 30 minutes my RAS PI OS is writing..

dubnemo, 2022-03-07:
@shreyashkashyap Hi, see the solution offer.  Not every reader/writer works, and the higher densities require SDHC compatible.  This was solved and closed.

Lucho-Vidal, 2022-07-11:
hola, tengo el mismo problema pero el USB es un disco duro de 320gb conectado por Carry de SATA a USB 3.2 de ADATA. Alguna sugerencia?