raspberrypi/rpi-imager #213: [Documentation] Advanced Options only known to work with RasPiOS images

DeeJay, 2021-04-20:
A documentation suggestion. Perhaps mention somewhere that the Advanced Options image customisation tool is probably only tested with Raspberry Pi RasPiOS images?

(Two instances to date in the Raspberry Pi forums of users trying to customise an Ubuntu image in this way and  being disappointed and confused when it results in a failed installation.)

lurch, 2021-04-20:
"failed installation"? I'd assume that anything that isn't RPiOS would just ignore the files that RPi Imager creates? :confused:

maxnet, 2021-04-20:
>"failed installation"? I'd assume that anything that isn't RPiOS would just ignore the files that RPi Imager creates?

Failed is expected.

We force systemd to start /boot/firstrun.sh
And if some OS decides to mount the FAT partition not at /boot but likes say /boot/EFI better, that do is fatal.

lurch, 2021-04-20:
Ahhhh. @maxnet I wonder if there should maybe be something in the JSON file that RPiImager reads, to know if the image it's flashing is "compatible" with the customisation options?   (although that wouldn't work if the user selects a RaspiOS image from their hard-drive, hmmm...)

maxnet, 2021-04-20:
>although that wouldn't work if the user selects a RaspiOS image from their hard-drive, hmmm...

Can have all kinds of heuristics e.g. based on issue.txt on FAT partitions.

But long term it may be better to add cloud-init support to RPI OS.
(Ubuntu does already support that).

ghollingworth, 2021-04-21:
Indeed that is what we discussed.  I think moving to cloud-init is the right solution here...

ghollingworth, 2021-04-21:
Sorry, didn't mean to close!

KonstantMan, 2021-09-15:
So apparently there are other people with the same issue as me and i understand that they might be technical limitations on what the program can set before flashing the OS. But the problem is there is no way to know before starting to use rpi - imager that's only fully compatible with Raspberry OS it would be very helpful for a new user to state clearly that the advanced settings only work with Raspberry OS

madfusker, 2021-10-31:
> A documentation suggestion. 

I just spent a couple hours on this one...  Instead of documentation, why not just remove the advanced options from being available when Ubuntu is selected.  Perhaps a dialog box when hitting ctrl-shift-x that says "Advance options for pi images only".

bryanpedini, 2021-12-09:
> > A documentation suggestion.
> 
> I just spent a couple hours on this one... Instead of documentation, why not [...]

Also document it please, I just ran across this issue and I had to resort to Google (instead of my default DuckDuckGo) by actually quoting "command from kernel command line" which brought me to https://github.com/raspberrypi/linux/issues/4458 which then searching on [raspberrypi/rpi-imager](https://github.com/raspberrypi/rpi-imager)'s issues from that issue's date on, pointed me to #273 which finally brought me here through #242...

Maybe some more SEO with a quick drafted article explaining that this error is normal and the reasons (we're not using RPiOS hence customizations through CTRL+SHIFT+X brick our installation), would be really, really needed! 😄

bryanpedini, 2021-12-09:
(also now that I mentioned all those issues here on GH, at least they all have a point-of-reference at the end of the page to click and quickly know what's going on... so thanks GitHub for already providing a small amout of SEO and alleviating the headake that figuring this issue out has been 😅)

SupineDread, 2022-01-24:
Hi guys 
I just disabled all the advanced options like this

[image]

And now is working as expected and using the default ubuntu user and password (ubuntu:ubuntu) I have access with ssh

cillian64, 2023-10-19:
With the built-in OS list we now only allow customisation to be applied to images which should support it.  But you can still attempt to apply customisation if you load a custom .img file and we have no way of knowing whether or not it will work.  I'd like to add a warning in this case to make it clear that it probably won't work if your custom image isn't rpi-os

tdewey-rpi, 2024-08-28:
Closing as resolved.

We don't present OS Customisation at all if you're using 'Use Custom' - but you _can_ apply customisations using the Raspberry Pi Imager CLI.

This is a deliberate posture - if you're using a custom image, you are expected to carry a certain amount of technical competence which allows you to drive CLI tools.

closed 2024-08-28 by tdewey-rpi

taylorkline, 2024-10-17:
> We don't present OS Customisation at all if you're using 'Use Custom' - but you _can_ apply customisations using the Raspberry Pi Imager CLI.

@tdewey-rpi Can you elaborate? I intentionally downloaded the imager version 1.9.0 for Windows to ensure I had a version newer than your comment, but I continue to be prompted for OS Customization settings for a Fedora 40 Server "Use custom" image I have selected in the GUI:

[image]

And, at the very least, the SSH configuration does not appear to work, so my attempt at a headless Fedora 40 Server configuration through the imager fails.

tdewey-rpi, 2024-10-18:
> @tdewey-rpi Can you elaborate? I intentionally downloaded the imager version 1.9.0 for Windows to ensure I had a version newer than your comment, but I continue to be prompted for OS Customization [...]

Sure - as covered on #941, it appears that OS customisation is still being offered in custom scenarios in spite of the intent that it shouldn't be.

I'm currently holding 1.9.2 back while I enforce the break.

taylorkline, 2024-10-18:
Thanks a lot!