raspberrypi/rpi-imager #214: >I can see the FAT32 partition but of course no firstrun.sh file.

GerBreOwn, 2021-04-22:
>I can see the FAT32 partition but of course no firstrun.sh file.

Ok.
Can you try if this one works better?

[imager-1.7beta.zip](https://github.com/raspberrypi/rpi-imager/files/6177569/imager-1.7beta.zip)

Waits up to 3 seconds checking if config.txt exists on the drive letter before proceeding to write the changes.

_Originally posted by @maxnet in https://github.com/raspberrypi/rpi-imager/issues/171#issuecomment-803591932_

GerBreOwn, 2021-04-22:
The exe file does NOT run on my Ubuntu Linux machine. Do you have a Linux version yet?

maxnet, 2021-04-22:
Our Ubuntu build was not affected by this issue.
Are you perhaps running the third-party snap?

Use this one instead: https://downloads.raspberrypi.org/imager/imager_latest_amd64.deb

closed 2021-04-22 by maxnet