raspberrypi/rpi-imager #215: using imager 1.6.1 on Ubuntu 20.10 to write Buster Lite image gives "unable to write firststart.sh" when using advanced options

GerBreOwn, 2021-04-22:
I see that there is a version 1.7 of Imager but it is only available for windoze. When will it be available for LINUX???
Thanks

maxnet, 2021-04-22:
1.7 beta was renamed to 1.6.2 because it did not really add any major chances.

maxnet, 2021-04-22:
Do make sure you use the .deb provided by raspberrypi.org.

The snap Canocial is pushing in the Ubuntu store is NOT from us, and is known broken.

closed 2021-04-22 by maxnet

lurch, 2021-04-22:
ping @waveform80 (who works for Canonical IIRC) in case he's able to "fix" that.

maxnet, 2021-04-22:
>ping @waveform80 (who works for Canonical IIRC) in case he's able to "fix" that.

Well, the related FAT mounting issue has already been open since September last year...
https://github.com/popey/imager-snap/issues/8

Would be better if they take down their build, if they do not plan to fix it.