raspberrypi/rpi-imager #217: Wrong progress on custom image selection

bkbilly, 2021-04-27:
I am using rpi-imager v1.6.1 and I have successfully installed the jetson-nano image which is around 14GB.
The problem is that the percentage goes way above 100%, I think it finished around 400%.

It's not an important issue and I know that this tool is not designed to install images for other platforms, but I thought I could raise the issue here because I like this tool.

maxnet, 2021-04-27:
Unable to reproduce.
Does not go above 99%

- Using Imager 1.6.1 on Ubuntu 20.04 amd64, with the standard .deb
- With the image from: https://developer.nvidia.com/jetson-nano-sd-card-image
- Feeding the jetson-nano-jp451-sd-card-image.zip file directly to Imager (not uncompressing it first).

What platform are you running Imager on?

bkbilly, 2021-04-27:
I am also using Ubuntu 20.04 amd64, but I have installed rpi-imager through snap.
I was trying to install it uncompressed as an img file.
Didn't notice that it supports zip files.

maxnet, 2021-04-27:
>I am also using Ubuntu 20.04 amd64, but I have installed rpi-imager through snap.

Please try again with the .deb (available on the releases page here, or on raspberrypi.org )

We have had multiple people report issues here that only occur with the snap.
If it is not reproducible with the .deb please report the issue here: https://github.com/popey/imager-snap/issues/

bkbilly, 2021-04-27:
Thanks for your help. I will try the deb, and since you've tried it, I am closing this issue and report any problem on the snap imager repository.

closed 2021-04-27 by bkbilly