raspberrypi/rpi-imager #219: [OS Request] Homebridge (JSON ready)

oznu, 2021-04-29:
Hello,

I'm a maintainer of the [Homebridge](https://github.com/homebridge/homebridge) project and would like to investigate the possibility of having the [Homebridge Raspberry Pi Image](https://github.com/homebridge/homebridge-raspbian-image) added to the *Other specific purpose OS* section of the RPI Imager tool.

I have created the relevant JSON:

```json
{
  "name": "Homebridge",
  "description": "Turn your Raspberry Pi into a HomeKit smart home bridge.",
  "icon": "https://user-images.githubusercontent.com/3979615/116509191-3c35f880-a906-11eb-9a7f-7cad7c2aa641.png",
  "subitems_url": "https://homebridge.io/rpi-image-repo.json"
}
```

As displayed under the **"Other specific purpose OS"** category:

[image]

OS List:

[image]

ghollingworth, 2021-05-05:
Done

closed 2021-05-05 by ghollingworth

ghollingworth, 2022-02-02:
Can you email me at gordon@ this organisation dot com