raspberrypi/rpi-imager #220: Windows: imager disappears immediately, following disconnect of secondary display

ComputerDoktor, 2021-05-04:
incomplete fix: commit https://github.com/raspberrypi/rpi-imager/commit/c08887f788bf1e67d246403d61dd5025e2b4765d 
as suggested by @gsh, here is my scenario
screen 2 above screen one. Start rpi-imager, move it to screen 2, click on OS selection, close.
resulting registry entry
[HKEY_CURRENT_USER\SOFTWARE\Raspberry Pi\Imager]
"x"=dword:00000254
"y"=dword:fffffb78   
disconnect screen 2
start rpi-manager, answer UAC, rpi-manager does not open, icon on taskbar cannot be restored to normal size without tricks or registry modification (deletion of rrpi-imager key)

ghollingworth, 2021-05-04:
Can reproduce... @maxnet can we have a look at why this is different to the previous issue...

Note, my previous workaround does still work:

1) Start rpi-imager
2) Press Alt-Space
3) Press and hold the down arrow until the window moves back to the display 1

maxnet, 2021-05-04:
Probably because we were only checking if x/y were larger than dimensions of primary display, and it is probably using negative coordinates instead if dragging screen 2 before screen 1.

Can you try if this one works better for you?

[imager-1.6.2.zip](https://github.com/raspberrypi/rpi-imager/files/6424039/imager-1.6.2.zip)

ghollingworth, 2021-05-05:
Fixes on my desk

ComputerDoktor, 2021-05-05:
fixes my scenario too

maxnet, 2021-05-26:
Fixed in 1.6.2

closed 2021-05-26 by maxnet