raspberrypi/rpi-imager #221: Add command-line support

kelson42, 2021-05-05:
We would like to use the imager instrumented by other scripts/softwares but it works only in with a GUI. Would be great to be able to write images in command-line only mode and non-interactively.

ghollingworth, 2021-05-05:
How would you suggest identifying the image to install?

Currently, the image list is generated from an internal .json file here:  

https://downloads.raspberrypi.org/os_list_imagingutility_v2.json

This then has links to externally generated .json files.  So to specify a particular image is difficult when the menu list is dynamic...

Gordon

lurch, 2021-05-05:
For "non-interactive mode" you could always rely on just `wget` and `dd` ? :shrug:

ghollingworth, 2021-05-05:
Well not really, rpi-imager does a lot more than just wget | dd

lurch, 2021-05-05:
> rpi-imager does a lot more than just wget | dd

Yeah I know - my comment was aimed at @kelson42 (i.e. as an alternative to command-line rpi-imager)

rpi-imager indeed does a lot more then `wget | dd`, but AFAIK it wasn't written to run in a "command-line only mode", so trying to squeeze RPi Imager's functionality (select image, select drive, display write progress, display verification progress, etc. etc.) would (probably) be a heck of a lot of work.

rgaudin, 2021-05-05:
What we're the most interested in is the writing part and the multiplatform-ness.
We are currently using etcher-cli but it's been abandoned a while ago.

Its CLI satisfies our needs:

```
Usage: etcher.js [options] <image>

Options:
  --help, -h     show help  [boolean]
  --version, -v  show version number  [boolean]
  --drive, -d    drive  [string]
  --check, -c    validate write  [boolean] [default: true]
  --yes, -y      confirm non-interactively  [boolean]
  --unmount, -u  unmount on success  [boolean] [default: true]

Examples:
  etcher.js raspberry-pi.img
  etcher.js --no-check raspberry-pi.img
  etcher.js -d /dev/disk2 ubuntu.iso
  etcher.js -d /dev/disk2 -y rpi.img

Exit codes:
  0 - Success
  1 - General Error
  2 - Validation Error
  3 - Cancelled
```

We used to choose it because etcher includes a lot of logic to work around common and not-so-common issues regarding SD card specs.
A quick look at https://github.com/raspberrypi/rpi-imager/blob/qml/imagewriter.cpp seems to indicate you don't do such magic yet you don't seem to face much problem though.

maxnet, 2021-05-05:
>Would be great to be able to write images in command-line only mode and **non-interactively**.
[...]
>Examples:
>  etcher.js raspberry-pi.img

So then it should guess which destination drive it should write to?
Sounds dangerous.

rgaudin, 2021-05-05:
> >Would be great to be able to write images in command-line only mode and **non-interactively**.
> 
> [...]
> 
> >Examples:
> 
> >  etcher.js raspberry-pi.img
> 
> 
> 
> So then it should guess which destination drive it should write to?
> 
> Sounds dangerous.

Not at all. You specify an input image and a destination drive

maxnet, 2021-05-05:
>AFAIK it wasn't written to run in a "command-line only mode", so trying to squeeze RPi Imager's functionality (select image, select drive, display write progress, display verification progress, etc. etc.) would
>(probably) be a heck of a lot of work.

Technically, the code does have proper separation between front-end and backend, and if you really want a CLI application, you should be able to write a minimal one in 20 lines of code among the lines of:

```
int main(int argc, char *argv[])
{
	QCoreApplication app(argc, argv);
	ImageWriter imageWriter;
	
	// Set file/URL
	imageWriter.setSrc(app.args[1]);
	// Set destination device
	imageWriter.setDst(app.args[2]);
	
	connect(&imageWriter, &ImageWriter::success, [=]() {
		qDebug() << "Success!";
		QCoreApplication::exit(0);
	});
	
	connect(&imageWriter, &ImageWriter::error, [=](QVariant msg) {
		qDebug() << "Error:" << msg;
		QCoreApplication::exit(1);
	});
	
	imageWriter.startWrite();
	// Main event loop
	app.exec();
}
```

(untested)

But the elephant in the room is and remains selecting the right destination drive.
Especially if you intend to run the code on end-user computers that are not yours (just a guess, based on rgaudin mentioning he likes the multiplatform-ness. Which is usually not something needed if it only needs to run on the single platform you use yourself).

rgaudin, 2021-05-05:
Thanks for the confirmation. We already have the device selection part in our GUI (we create images) and we already pass device path to etcher (using a different syntax per platform).
We'll look into that cli POC.

maxnet, 2021-05-06:
Checked in a "--cli [url] [device]" option that may (or may not) do what you want.

closed 2021-05-06 by maxnet

kelson42, 2021-05-06:
@maxnet thx so much

rgaudin, 2021-05-06:
Thank you ; gonna test it right away :)

lurch, 2021-05-06:
...and to help with determining the `[url]` part, I knocked up a cool little command-line python script this evening which you can use to navigate the same JSON-hierarchy that RPi Imager uses, and once you 'select' an image, it gives you the URL that the image can be downloaded from.
https://gist.github.com/lurch/e675ccaf691cc6888d81199276994c7e

@procount and @XECDesign might find this script interesting / useful too? :slightly_smiling_face:

rgaudin, 2021-05-07:
Thank you @lurch ; I've tested it and it works fine. That's exactly what we needed. Only tested on macOS though. 

Only concern so far is that you've been good citizen in properly requiring device permission from the user but it's not checking for prior authorization such as running via sudo. That would be a problem for our unattended tool.
It seems Linux version is OK though as it tries to open the device first then only fallback to udisks. Windows appears to just open the device ; I guess the whole app requires elevation earlier.

Quick note ; compilation fails on macOS at the moment

```
[ 97%] Building CXX object CMakeFiles/rpi-imager.dir/mac/macfile.cpp.o
/Users/reg/src/rpi-imager/mac/macfile.cpp:87:32: error: use of undeclared identifier 'errno'
        } while (size == -1 && errno == EINTR);
                               ^
/Users/reg/src/rpi-imager/mac/macfile.cpp:87:41: error: use of undeclared identifier 'EINTR'
        } while (size == -1 && errno == EINTR);
                                        ^
/Users/reg/src/rpi-imager/mac/macfile.cpp:108:32: error: use of undeclared identifier 'errno'
        } while (wpid == -1 && errno == EINTR);
                               ^
/Users/reg/src/rpi-imager/mac/macfile.cpp:108:41: error: use of undeclared identifier 'EINTR'
        } while (wpid == -1 && errno == EINTR);
                                        ^
4 errors generated.
```

Gets fixed by including `errno.h`.

``` patch
diff --git i/mac/macfile.cpp w/mac/macfile.cpp
index 91529db..5ccc949 100644
--- i/mac/macfile.cpp
+++ w/mac/macfile.cpp
@@ -6,6 +6,7 @@
 #include "macfile.h"
 #include <unistd.h>
 #include <fcntl.h>
+#include <errno.h>
 #include <sys/socket.h>
 #include <security/Authorization.h>
 #include <QDebug>
```

maxnet, 2021-05-07:
>Quick note ; compilation fails on macOS at the moment

Does work with my version of xcode.
But will add the errno.h.

commit 16bee622 (2021-05-06): Basic CLI support

Closes #221
commit 62e9969a (2021-05-06): Basic CLI support

Closes #221