raspberrypi/rpi-imager #225: [OS Request] HOOBS (JSON ready)

mkellsy, 2021-05-18:
Hi,

I am a co-creator of [HOOBS](https://hoobs.org/). We are inquiring about adding our Raspberry Pi image to Other specific purpose OS section of the RPI Imager tool.

I have created the relevant JSON:
```json
{
    "name": "Homebridge",
    "description": "Turn your Raspberry Pi into a HomeKit smart home bridge.",
    "icon": "https://repo.hoobs.org/hoobs-icon.png",
    "subitems_url": "https://repo.hoobs.org/subitems.json"
}
```

Thank You

> The RPI Imager tool is sweet!

ghollingworth, 2021-05-28:
Looking at the icon it looks exactly the same as the Homebridge icon (although a different colour), are these the same software?

mkellsy, 2021-05-28:
We are related.

closed 2021-09-08 by mkellsy