raspberrypi/rpi-imager #226: [OS Request] LynxCI (JSON ready)

getlynx, 2021-05-20:
Hello,

I am the founder and Team Lead for the Lynx Cryptocurrency Project. Our slightly customized Raspberry Pi OS (32) image is a preconfigured and secured Lynx mining device (and wallet). Heavily tested with hundreds of international installs. Looking to make setups even easier for our global community of enthusiasts. We are inquiring about adding our Raspberry Pi image to the "Other specific purpose OS" section of the RPI Imager tool.

The respective rpi-imager.json file has been created and published:

		{
			"name": "LynxCI",
			"description": "Be mining Lynx cryptocurrency in 20 minutes. Automated & lots of help.",
			"icon": "https://github.com/getlynx/LynxCI/releases/download/v27-ISO/LYNX_3COLOR_40square.png",
			"subitems_url": "https://github.com/getlynx/LynxCI/releases/download/v27-ISO/rpi-imager.json"
		}

Thx
-ben

RE: https://github.com/getlynx/LynxCI & https://getlynx.io/

ghollingworth, 2021-05-28:
Think this would be better as an debian package

closed 2021-05-28 by ghollingworth

getlynx, 2021-05-28:
@ghollingworth Our developers have been researching a debian package as well. It's on the roadmap for 2022 if we can figure out how to include some of the OS level modifications we include with this IMG. It really converts the OS into a dedicated device, we recommend that users NOT install it with other packages pre-installed thus the 'specific purpose OS' classification request. Our project is about ease of access and quickly getting a node running for non-techies and enthusiasts. Using the IMG we created, our community can be running a node without ever even logging into the device. Flash the card, plugin in the eth cable, plugin in the power, done.

With the debian package option, the user flashes the card with RaspbianOS Lite, logs in as root and watches the 1 line installer (https://github.com/getlynx/LynxCI) outputs. We have already lost several users with the more complex steps. Granted, for a nerd like me, it seems silly, but our UAT feedback has shown repeatedly that users attempting to get involved drops off with the more steps involved.

Respectfully, can I request a reconsideration of this inclusion? -ben