raspberrypi/rpi-imager #227: Run time dependencies (Fedora)

kees-closed, 2021-05-22:
Hi,

The [build dependencies for Fedora](https://github.com/raspberrypi/rpi-imager#fedorarhelcentos-linux) are well documented. But it does not mention the runtime dependencies. On a bare installation of Fedora the package `qt5-qtquickcontrols2` is needed to start the application and `qt5-qtsvg` to fix the SVG errors at startup. I would guess that the following are also needed: `libcurl`, `openssl`, `libarchive`, `qt5-qtbase`. But that's just a wild guess based on the build depends, which are then the devel packages. Could someone please confirm what exactly is needed? The [Debian package](https://github.com/raspberrypi/rpi-imager/blob/qml/debian/control) only recommends the `udisks2` package. That for sure it not enough. Thanks!

maxnet, 2021-05-22:
>The Debian package only recommends the udisks2 package. That for sure it not enough.

"Recommends" means optional dependencies.
(udisks2 do is needed if you want to be able to run rpi-imager as a regular user. But you do could run Imager without udisks2 if started as root)

You will find the required runtime dependencies in the control file under "Depends"

On github it is:

``
Depends: ${shlibs:Depends}, ${misc:Depends},  qml-module-qtquick2, qml-module-qtquick-controls2, qml-module-qtquick-layouts, qml-module-qtquick-templates2, qml-module-qtquick-window2, qml-module-qtgraphicaleffects, dosfstools, fdisk 
``

${shlibs:Depends} do is a magic construct that automatically adds the packages of all shared libraries that the main executable directly links to, to the list.
Not sure if there is a Fedora equivalent for that.

It expands to:

``
Depends: libarchive13 (>= 3.0.4), libc6 (>= 2.14), libcurl4 (>= 7.16.2), libgcc1 (>= 1:3.0), libqt5core5a (>= 5.11.0~rc1), libqt5dbus5 (>= 5.0.2), libqt5gui5 (>= 5.2.0), libqt5network5 (>= 5.0.2), libqt5qml5 (>= 5.1.0), libqt5quick5 (>= 5.0.2), libqt5svg5 (>= 5.6.0~beta), libqt5widgets5 (>= 5.0.2), libssl1.1 (>= 1.1.0), libstdc++6 (>= 5.2), qml-module-qtquick2, qml-module-qtquick-controls2, qml-module-qtquick-layouts, qml-module-qtquick-templates2, qml-module-qtquick-window2, qml-module-qtgraphicaleffects, dosfstools, fdisk
Recommends: udisks2
``

kees-closed, 2021-05-22:
Thanks, got it. In RPM it's `Requires` which is equivalent to `Depends` and `Recommends` in RPM is the same as in Debian. I'll update the RPM spec file with the deb file as a guide. But I can already see that the package names are not the same. It would therefore be nice to have the runtime dependencies listed for Fedora/RedHat/SUSE systems as well. Otherwise packagers have to do educated guesses, which may work, but may also result in "bugs" when running the tool without the needed dependencies.

kees-closed, 2021-05-24:
Updates are on their way: https://bugzilla.redhat.com/show_bug.cgi?id=1964130

closed 2021-05-24 by kees-closed