raspberrypi/rpi-imager #228: [Feature  Request] - Add a SDCard backup function, please.

jedi101, 2021-05-24:
I know the imager is designed to be as entry user friendly as possible, but do you mind adding an one click solution to backup the current Sdcard? (Maybe at least inside the advanced user menu)

Use Case:

* Sometimes the user just want to try out a new os from the list of operatring systems to choose from, but what if you don't have 12 or more spare sdcards for that on hand? Than you would have to always overwrite your beloved and already configured raspberry pi distribution entirely from the sdcard before you can try out something else. Thats a bit frustrating.

If there were such a backup option for the currently stored distribution on the sdcard this would provide the more inexpirinced user a save and easy way to also try out new distros without worring in loosing their already configured os on their pi.

That would be absolutely awesome.

ghollingworth, 2021-05-25:
There is no reliable way of backing up an image from an SD card on Windows.  You can copy the contents to an image, (i.e. you should use win32DiskImager), but if you then copy this to another SD card there is a real possibility the Linux kernel will not boot correctly.

This is because the Linux kernel doesn't like partitions that go off the end of a disk.  Since all SD cards are not the same size. Even if it says 32G on the side that doesn't mean you actually get 34,359,738,368 bytes available on the card, because some number of blocks will be marked as 'bad' so it's always less than that.  So if you copy the bytes from a card which is 34,359,226,368 onto a card that only has 34,357,690,368 bytes available the end of the partition will go off the end of the card.

Because of this, it means you have to do special manipulation of the filesystem which you can't do using a Windows operating system (Windows doesn't understand the underlying filesystem to make manipulations to it).

I would suggest instead using "SD Card Copier" on the Raspberry Pi OS to make a copy of the SD card instead, this works by creating a new filesystem on the SD card and copying the individual files across

closed 2021-05-25 by ghollingworth

lurch, 2021-05-25:
> but what if you don't have 12 or more spare sdcards for that on hand?

The way I "solve" this problem is to have my "beloved and already configured raspberry pi distribution" on one card which I never overwrite, and then have a "spare" SD card which I use for experimenting with different OSes - and because I know that this is my "experimentation SD card" I know that I'll probably overwrite it with some other OS in the future, so I don't spend too much time making extensive customisations to it (i.e. I treat the entire card as "temporary").

Only needs 2 SD cards, not 12 :wink:

jesus2099, 2021-12-23:
These 2 answers, [_SD Card Copier_](#issuecomment-847550069) and [_experimentation SD card_](#issuecomment-847654766), only apply when you have a spare SD card, which is not the case of the OP.

I have the same need. I don't want to have a spare SD card lying around, just for testing Raspbian 11, for instance.