raspberrypi/rpi-imager #229: Invalid wpa_supplicant if password contains double quote ( " )

matthew1471, 2021-05-25:
Setting a wireless password that contains a " creates a wpa_supplicant which is invalid. The correct way to specify these passwords is by using wpa_passphrase (or the online version at https://www.wireshark.org/tools/wpa-psk.html)

firstrun.sh

```
...
cat >/etc/wpa_supplicant/wpa_supplicant.conf <<WPAEOF
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
ap_scan=1

update_config=1
network={
	ssid="MyWiFi"
	psk="Ilovefishes"Ido!"
}

WPAEOF
...
```

maxnet, 2021-05-25:
Already fixed in #206
(Just no new version released since)

closed 2021-05-25 by maxnet

matthew1471, 2021-05-25:
Doesn't say fixed in linked issue.. what version number is this fixed in please?

maxnet, 2021-05-25:
>what version number is this fixed in please?

Fixed in source code in git commit b57da413a62b6c175807278dc0996e11372399de

As I said though, there has not been any new version publicly released since.
So you will either have to build from source or wait.

lurch, 2021-05-25:
It's fixed in the source-code https://github.com/raspberrypi/rpi-imager/commit/b57da413a62b6c175807278dc0996e11372399de but there's not yet been a "release build" of RPi Imager which includes those changes. If you're desperate to pick up this fix in the short-term (before the next release build) you'll need to build your own version of RPi Imager from source.

matthew1471, 2021-05-25:
I understand - thank you! :)

lurch, 2021-05-26:
1.6.2 has just been officially released :smiley: