raspberrypi/rpi-imager #231: -Reconsideration [OS Request] LynxCI (JSON ready)

getlynx, 2021-06-01:
@ghollingworth Our developers have been researching an ubuntu/debian package as well. It's on the roadmap for 2022 if we can figure out how to include some of the OS level modifications we include with this IMG. It really converts the OS into a dedicated device (adjusts SWAP, locks user accounts, complete firewall setup, etc), we recommend that users NOT install it with other packages as it might create a conflict or impact existing services pre-installed thus the 'specific purpose OS' classification request. Our project is about ease of access and quickly getting a node running for non-techies and enthusiasts. Using the IMG we created, our community can be running a node without ever even logging into the device. Flash the card, plugin in the eth cable, plugin in the power, done.

With the ubuntu/debian package option, the user flashes the card with RaspbianOS Lite, logs in as root adds the PPA link and installs. For security reasons the design of our system doesn't include an upgrade process (precious data cold be lost) - only new creations so a PPA isn't really an ideal installation process. Users also could run our 1 line installer (https://github.com/getlynx/LynxCI) and watch the outputs. We have already lost several users with the more complex steps. Granted, for a nerd like me, it seems silly, but our UAT feedback has shown repeatedly that users attempting to get involved drops off with the more steps involved.

Respectfully, can I request a reconsideration of this inclusion? -ben

ghollingworth, 2021-06-02:
In the end, we don't think rpi-imager is the right place for this type of image.  You can instead provide people with an image on your website and point them at rpi-imager to install it to an SD card.

closed 2021-06-02 by ghollingworth

getlynx, 2021-06-02:
Thank you @ghollingworth for the second look. Much appreciated.