raspberrypi/rpi-imager #237: rpi-imager-1.62 fails to write to sd card

juanitotc, 2021-06-07:
rpi-imager-1.62 compiled from source fails to write to sd card with:

```
Telemetry done. cURL status code = 0
sfdisk: "Checking that no-one is using this disk right now ... OK\n\nDisk /dev/mmcblk0: 14.45 GiB, 15510536192 bytes, 30294016 sectors\nUnits: sectors of 1 * 512 = 512 bytes\nSector size (logical/physical): 512 bytes / 512 bytes\nI/O size (minimum/optimal): 512 bytes / 512 bytes\nDisklabel type: dos\nDisk identifier: 0x2ddfd8be\n\nOld situation:\n\nDevice         Boot Start      End  Sectors  Size Id Type\n/dev/mmcblk0p1       8192 30294015 30285824 14.4G  e W95 FAT16 (LBA)\n\n>>> Created a new DOS disklabel with disk identifier 0xd35f0ab7.\n/dev/mmcblk0p1: Created a new partition 1 of type 'W95 FAT16 (LBA)' and of size 14.4 GiB.\nPartition #1 contains a vfat signature.\n/dev/mmcblk0p2: Ignoring partition.\n/dev/mmcblk0p3: Ignoring partition.\n/dev/mmcblk0p4: Ignoring partition.\n/dev/mmcblk0p5: Done.\n\nNew situation:\nDisklabel type: dos\nDisk identifier: 0xd35f0ab7\n\nDevice         Boot Start      End  Sectors  Size Id Type\n/dev/mmcblk0p1       8192 30294015 30285824 14.4G  e W95 FAT16 (LBA)\n\nThe partition table has been altered.\nCalling ioctl() to re-read partition table.\nSyncing disks.\n"
mkfs.fat: "mkfs.fat: unable to open /dev/mmcblk0p1: No such file or directory\nmkfs.fat 4.2 (2021-01-31)\n"
```

..but /dev/mmcblk0p1 is present

maxnet, 2021-06-07:
I am guessing you are executing Imager as root.
Does it work better if you run it as normal user? (in which case it will follow a slightly different code path in which it outsources partitioning and formatting to udisks2 instead of calling sfdisk/mkfs.fat itself).

Could be a race condition where the /dev/mmcblk0p2 device node is not created yet by the time it tries.
Although I am not sure why that would be the case, as it does re-read the partion table, and assuming /dev is handled by devtmpfs the entry should be there instantly.

juanitotc, 2021-06-07:
You're correct, I was running as root.

I tried with udisks, which is quite possible not configured correctly, and get (as user):

```
sfdisk: "Checking that no-one is using this disk right now ... FAILED\n\nThis disk is currently in use - repartitioning is probably a bad idea.\nUmount all file systems, and swapoff all swap partitions on this disk.\nUse the --no-reread flag to suppress this check.\n\nsfdisk: Use the --force flag to overrule all checks.\n"
Telemetry done. cURL status code = 0
```

The sd card is not mounted

lurch, 2021-06-07:
Which distro and version are you running?

juanitotc, 2021-06-07:
Tinycorelinux CorePure64 12.x

maxnet, 2021-06-07:
partprobe (usually part of gparted) do is installed and in path?
As that is executed between sfdisk and mkfs.fat if running as root.

juanitotc, 2021-06-07:
I loaded parted and re-run rpi-imager as root, but the same error remains.

I also ran rpi-imager as user in the gnome environment so that udisks would be started correctly, but again the same error remains.

```
$ ps aux | grep udisks
 8132 tc       /usr/local/lib/gvfs/gvfs-udisks2-volume-monitor
 8140 root     /usr/local/lib/udisks/udisks2/udisksd
```

```
$ udisksctl status
MODEL                     REVISION  SERIAL               DEVICE
--------------------------------------------------------------------------
SAMSUNG SSD PM851 mSATA 256GB EXT48DQ0  S1EVNYAF177947       sda     
Y016U                               0xdad2c06f           mmcblk0 
Sony Storage Media        PMAP      9C070D66656C478C78   sdb
```

lurch, 2021-06-07:
I'm guessing the fact that your SD card shows up as `mmcblk0` means that it's an internal slot in a laptop? _Sometimes_ the drivers for those aren't as robust as they should be. (And perhaps Tinycorelinux includes fewer drivers than more full-fat distros?) Do you see different behaviour if you plug in your SD card using a USB SD card-reader?

maxnet, 2021-06-07:
>I loaded parted and re-run rpi-imager as root, but the same error remains.

For the "/dev/mmcblk0p1: No such file or directory" error when running as root, while it did was available later when you checked manually, you can try adding a delay here: https://github.com/raspberrypi/rpi-imager/blob/b57da413a62b6c175807278dc0996e11372399de/driveformatthread.cpp#L190

like:

```
for (int try = 0; try < 100; try++)
{
	if (QFile::exists(fatpartition))
		break;
	
	qDebug() << fatpartition << "not available yet.";
	QThread::msleep(100);
}
```

But it really should not be necessary.

juanitotc, 2021-06-07:
Yes, it's a slot in a laptop - I don't have a usb sd card reader to try.

Note that I've prepared many sd cards for Tinycorelinux piCore/piCore64 using the same sd card interface.

juanitotc, 2021-06-07:
I'll try the delay suggestion tomorrow

maxnet, 2021-06-07:
BTW if you do "ps aux |grep dev" it does says some udevd or systemd-udevd is running, and not some alternative device manager like mdev?

(As the latter also has a known problem that when new device nodes appear it first deletes the /dev/name-of-device created by devtmpfs, and then recreates it with the permissions mdev likes afterwards.
If a program tries to access the device between the moment it deleted it, and the moment it recreated that is also a problem)

juanitotc, 2021-06-08:
As above:

```
$ ps aux | grep udisks
 8132 tc       /usr/local/lib/gvfs/gvfs-udisks2-volume-monitor
 8140 root     /usr/local/lib/udisks/udisks2/udisksd
```

Adding the delay to driveformatthread.cpp fixed things for running rpi-imager as root - thanks

Note that I had to use "i" instead of "try" to avoid a compiler error "

```
$ sudo rpi-imager
...
sfdisk: "Checking that no-one is using this disk right now ... OK\n\nDisk /dev/mmcblk0: 14.45 GiB, 15510536192 bytes, 30294016 sectors\nUnits: sectors of 1 * 512 = 512 bytes\nSector size (logical/physical): 512 bytes / 512 bytes\nI/O size (minimum/optimal): 512 bytes / 512 bytes\nDisklabel type: dos\nDisk identifier: 0xb58cb42b\n\nOld situation:\n\nDevice         Boot Start      End  Sectors  Size Id Type\n/dev/mmcblk0p1       8192 30294015 30285824 14.4G  e W95 FAT16 (LBA)\n\n>>> Created a new DOS disklabel with disk identifier 0x31113e2e.\n/dev/mmcblk0p1: Created a new partition 1 of type 'W95 FAT16 (LBA)' and of size 14.4 GiB.\nPartition #1 contains a vfat signature.\n/dev/mmcblk0p2: Ignoring partition.\n/dev/mmcblk0p3: Ignoring partition.\n/dev/mmcblk0p4: Ignoring partition.\n/dev/mmcblk0p5: Done.\n\nNew situation:\nDisklabel type: dos\nDisk identifier: 0x31113e2e\n\nDevice         Boot Start      End  Sectors  Size Id Type\n/dev/mmcblk0p1       8192 30294015 30285824 14.4G  e W95 FAT16 (LBA)\n\nThe partition table has been altered.\nCalling ioctl() to re-read partition table.\nSyncing disks.\n"
"/dev/mmcblk0p1" not available yet.
mkfs.fat: "mkfs.fat 4.2 (2021-01-31)\n"
Image URL: "https://github.com/raspberrypi/rpi-eeprom/releases/download/v2021.04.29-138a1/rpi-boot-eeprom-recovery-2021-04-29-vl805-000138a1.zip"
...
```

maxnet, 2021-06-08:
>As above:
>
>$ ps aux | grep udisks
> 8132 tc       /usr/local/lib/gvfs/gvfs-udisks2-volume-monitor
> 8140 root     /usr/local/lib/udisks/udisks2/udisksd

Was not interested in udisks, but if udev or something alternative like mdev was used by your Linux distro.
Those handle the device nodes living in /dev

juanitotc, 2021-06-09:
Ah sorry, udisks had been mentioned earlier..

```
$ ps aux |grep dev
   79 root     [devfreq_wq]
  161 root     /sbin/udevd --daemon
 2885 tc       grep dev
25863 root     /sbin/udevd --daemon
25864 root     /sbin/udevd --daemon
```

commit 2e5cc750 (2021-11-18): Linux (as root): FAT32 format: wait for FAT partition after sfdisk

The FAT partition /dev node is supposed to be already available
after running partprobe, but since there is a user report that
is not always the case, double check and delay if necessary.

Closes #237
closed 2021-11-18 by maxnet