raspberrypi/rpi-imager #240: [Feature Request] user added images

CAProjects, 2021-06-11:
It would be nice if there was a way for us users to add our own OS images that is stored on out own PC and add it to the menu or may be have a community category for us to be able to contribute OSs that do work on the Raspberry Pi. 

I know we can just do custom but would be nice to have them listed in the app than having to browse though your drives/network all the time

lurch, 2021-06-11:
https://github.com/raspberrypi/rpi-imager#custom-repository ?

CAProjects, 2021-06-12:
Its not what I was meaning, I was asking for a way to add local custom images to the Imager to quickly access them instead of having to navigate though directories all the time

tdewey-rpi, 2024-02-06:
Closing as won't address. The link above provides the mechanism for the OP to achieve their goal - they just need to create a custom repo JSON file with their desired images. Custom flashing support is also provided for users who don't want to create a custom JSON file.

closed 2024-02-06 by tdewey-rpi