raspberrypi/rpi-imager #241: Improvement - Changes on main GUI layout

bovirus, 2021-06-14:
Current layout

[image]

Proposal of new layout

[image]

**Changes**

- Move Write in bottom centered on wdth
- Enlarge button width for "S.O. name/ file name"
- Enlarge button width for "Select SD card /SD card name"
- Change "Operative system" text in "Operative system / file name"

ghollingworth, 2021-06-14:
Thanks we'll take it into consideration.

closed 2021-06-14 by ghollingworth

bovirus, 2021-06-14:
@ghollingworth

Pelase don't close it.
It's a pending request.
Leave it open. Thanks. KR.