raspberrypi/rpi-imager #243: [Feature Request] Add static IPs and file uploading for public key file

MohamedAliRashad, 2021-06-23:
Those are the two main features, i wish they were around in rpi-imager.
And maybe a button for advanced settings.

tdewey-rpi, 2024-02-06:
Closing as completed:

Static IP: Feature rejected. The better way to do this is to ask your DHCP server to provide a static lease.
Public Keyfile: This is covered by the authorized_keys customisation
OS Customisation button: This is covered in >1.8.1 by a mandatory confirmation step when you select an OS that supports customisation

closed 2024-02-06 by tdewey-rpi

notthetup, 2024-10-25:
> Static IP: Feature rejected. The better way to do this is to ask your DHCP server to provide a static lease.

What if the RPi is being used in a setup without a DHCP server? I think this would be super valuable for CM4 based system.

tdewey-rpi, 2024-10-25:
> > Static IP: Feature rejected. The better way to do this is to ask your DHCP server to provide a static lease.
> 
> What if the RPi is being used in a network without a DHCP server? I think this would be super valuable for CM4 based system.

This is not a good enough justification to offset the potential harm and resolution difficulty of misconfiguration.

In such an environment, I'd expect the user to be able to configure a DHCP server with static leases.

lurch, 2024-10-28:
...or to be a sufficiently advanced user that they know how to configure this manually themselves, without requiring help from Raspberry Pi Imager :wink: