raspberrypi/rpi-imager #244: Image Verification failed.

youaftouh, 2021-06-29:
Hi,
I just wanted to let you know that with rpi-imager, it didn't work for me.

I have RPI 3 model B, Windows 10.
I tried both default online image download and manual download by using custom image.
Also tried different recommended versions of the PRI images (with desktop, lite and full)
After writing and verifying, It says Verification failed.

Using win32 disk imager, it worked well.

lurch, 2021-06-29:
> Using win32 disk imager, it worked well.

Note that Win32DiskImager _only_ writes the image, it doesn't verify what was written (i.e. it's _possible_ for Win32DiskImager to write corrupted data).

Is it possible that your SD card is faulty, and that's why verification is always failing? Do you have a different SD card you could try?

youaftouh, 2021-06-29:
It's a brand new SD Card.
Every time it finishes writing on RPI Imager, I try it on the Raspberry with HDMI connected, shows nothing.
Until I try after writing the image using Win32DiskImager, it boots and works fine.

ghollingworth, 2021-06-29:
Brand new SD cards can (and have) had failed sectors in them.

RPi-Imager doesn't write the first sector until it has finished writing and verifying the rest of the card (unless you cancel the verification, in which case it will write the first sector.

So it's possible you've got a card with a bad sector in it which isn't verifying correctly.  To check this when it gets to the verifying stage hit the cancel verification button and then try again...

Note, it'll work the same as it does on  Win32DiskImager, but you will still have a bad SD card...

To check this (if you have the tools available) you could write to the card with Win32DiskImager, then read it back to a file and compare the file with the original file (there are some binary comparison tools available on the web)

Gordon

youaftouh, 2021-06-29:
This is a possible explanation, Thanks!
It's fine as long as it works for everyone, it could be an edge case with my SD card (having bad sectors..)