raspberrypi/rpi-imager #245: Allow installing RaspberryMatic

jens-maus, 2021-07-04:
Opening this issue according to https://www.raspberrypi.org/blog/raspberry-pi-imager-imaging-utility/#comment-1523756

[RaspberryMatic](https://github.com/jens-maus/RaspberryMatic) is an open source buildroot-based Linux OS alternative for running a full-fledged HomeMatic/homematicIP IoT central on RaspberryPi hardware. 

We provide operating system images which turn a RaspberryPi into a so-called HomeMatic central unit (CCU) for configuring and using HomeMatic/homematicIP IoT devices developed and distributed by the eQ3/ELV company group with a fairly large user group in Germany and Europe. While the HomeMatic vendor sells/provides own Hardware, our OS alternative provides a free and better alternative with more features and uptodate LinuxOS components. We support all variants of the Raspberry Pi starting from RaspberryPi1 till RaspberryPi4 including all variants of Compute Modules.

It would be great if RaspberryMatic would be available as an OS option in the Raspberry Pi Imager.

Our OS can be found here: https://github.com/jens-maus/RaspberryMatic/releases

By having studied how other operating systems like OctoPi or Home Assistant have provided necessary information, I have generated a similar `json` file in our public repository with the necessary information that rpi-image seems to require. By manually running `rpi-imager --repo https://raw.githubusercontent.com/jens-maus/RaspberryMatic/master/release/rpi-imager.json` the rpi-imager correctly picks up the necessary information and displays the following valid GUI:
[image]
Similar to having added Home Assistant under `Other specific purpose OS` I would like to request to add the following json passages to the public json `rpi-imager` is loading per default so that RaspberryMatic will be available to all `rpi-imager` users:

```json
                {
                    "name": "RaspberryMatic",
                    "description": "Lightweight Linux OS for running a HomeMatic/homematicIP IoT central.",
                    "icon": "https://raw.githubusercontent.com/jens-maus/RaspberryMatic/master/release/rpi-imager.png",
                    "subitems_url": "https://raw.githubusercontent.com/jens-maus/RaspberryMatic/master/release/rpi-imager.json"
                },
```

Thanks in advance for considering the addition of RaspberryMatic to the `Other specific purpose OS` section of `rpi-imager`

ghollingworth, 2021-07-05:
I can't add this while you are using the Raspberry Pi trademark in your logo.  You'll first need to change that (also I note on your website you use the Raspberry Pi logo).  Please read the trademark rules.

jens-maus, 2021-07-05:
@ghollingworth I see, did not really know that using the RaspberryPi logo in a pure non-commercial open source project is of any concern or poses a problem. I indeed was under the impression that the Pi logo is somewhat free to use for open source projects as it IMHO poses more free advertising. As said, RaspberryMatic is a 100% non-commercial open source project.

ghollingworth, 2021-07-05:
https://www.raspberrypi.org/trademark-rules/

The rules are clear, the logo can only be used by Raspberry Pi or in connection with the sale of Raspberry Pi products.

jens-maus, 2021-07-05:
@ghollingworth Ok, that's clear now. Thanks. I just sent you a private email on that matter to discuss how we can proceed here then and to clear things up before I am going to redesign the logo. Hope you received that mail.

jens-maus, 2021-07-09:
@ghollingworth As you seem to be quite busy, I have taken the road to redesign the RaspberryMatic logo to not violate your Raspberry Pi logo trademark. See here for the new rpi-imager screenshot with the new logo:

[image]

In addition, I have also changed the used images on the GitHub project pages as well as on our homepage to use the new logo.

Thus, would be great if you could please reconsider + adding RaspberryMatic as installable operating system images to rpi-imager. Or what else can I do to help you adding RaspberryMatic to rpi-imager? And sorry again for having used the Raspberry Pi trademark logo in first place - this was a clear misunderstanding on my side!

jens-maus, 2021-07-15:
@ghollingworth @maxnet Don't want to bother you once again, but what else can we do (we even changed our logo upon your request) to get our beloved "RaspberryMatic" added to the top-level http://downloads.raspberrypi.org/os_list_imagingutility_v2.json so that it will be available to standard rpi-imager users?

friedpa, 2021-07-15:
@ghollingworth @maxnet Please add "RaspberryMatic"

Expalmi, 2021-07-15:
@ghollingworth @maxnet It would be nice to see "RaspberryMatic" on your list

HMNutzer, 2021-07-15:
@ghollingworth @maxnet Please add RaspberryMatic by jens-maus

Franzens, 2021-07-15:
@ghollingworth @maxnet Please add RaspberryMatic by jens-maus to your list

toto79, 2021-07-16:
+1 for adding RaspberryMatic to your list

romeocontrol, 2021-07-17:
Open source is obviously not as easy as a one-man project imagines.
Nevertheless, the inclusion in the list in rpi-imager would be a support for this project, which carries the open source idea to a new field.
Especially since the maintainer, with your help, is making every effort to clear the legal hurdles.

tigger30926, 2021-07-17:
@ghollingworth @maxnet Please add "RaspberryMatic" to the list. That is a very nice Software. Thank you

ghollingworth, 2022-02-02:
Added

closed 2022-02-02 by ghollingworth