raspberrypi/rpi-imager #247: rpi-manager can't run  core dumped

asyba, 2021-07-07:
Ubuntu 18.04.5 LTS
installed with snap install rpi-imager

```
Qt: Session management error: Could not open network socket
Fontconfig warning: "/etc/fonts/conf.avail/53-monospace-lcd-filter.conf", line 10: Having multiple values in <test> isn't supported and may not work as expected
propsReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.369\" (uid=1000 pid=12323 comm=\"/snap/rpi-imager/184/usr/local/bin/rpi-imager \" label=\"snap.rpi-imager.rpi-imager (enforce)\") interface=\"org.freedesktop.DBus.Properties\" member=\"GetAll\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1268 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
nmReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.369\" (uid=1000 pid=12323 comm=\"/snap/rpi-imager/184/usr/local/bin/rpi-imager \" label=\"snap.rpi-imager.rpi-imager (enforce)\") interface=\"org.freedesktop.NetworkManager\" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1268 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
"Object path cannot be empty"
[1]    12323 segmentation fault (core dumped)  rpi-imager

```

lurch, 2021-07-07:
I think problems with the snap version need to be reported at https://github.com/popey/imager-snap/issues

But see also #197

asyba, 2021-07-07:
> I think problems with the snap version need to be reported at https://github.com/popey/imager-snap/issues
> 
> But see also #197

with this 

`LIBGL_ALWAYS_SOFTWARE=1 rpi-imager`

it work.

closed 2021-07-07 by asyba

maxnet, 2021-07-08:
>LIBGL_ALWAYS_SOFTWARE=1 rpi-imager

Perhaps because Ubuntu does not have the right graphic drivers for your GPU inside their snap container...
That it is the problem with containers, they often lack essential files specific to your system...