raspberrypi/rpi-imager #248: Add a option, to reset partitions of sd cards, that have been flashed with another application

dave90dave, 2021-07-14:
Hi,

I often flash a sd card with a custom firware.
When I want to format the sd card with Raspberry pi flasher, it does not reset the partitions.
It just formats the main (boot) partition, while retaining other hidden partitions.

Could there a option be added to reset the full partition table of a SD card? (that removes all partitions first, en thereafter formats the whole card (instead of formatting only the first partition that it finds)

Thanks for the support

Dave

maxnet, 2021-07-14:
The "erase / format as fat32" option do is supposed to repartition it as well.

What platform are you running Imager on, and how did you verify exactly other partitions are still there?

lurch, 2021-07-14:
I've no idea if it's still the case, and I can't remember the exact details (I can't even remember which OS it was), but I remember that at some point in the past I had to physically remove and then re-insert the SD-card to get the OS to re-read the partition table?
(Or perhaps I'm mis-remembering and getting confused with something else :shrug: )

dave90dave, 2021-07-16:
Yes, that is right.

After removal of the SD card, it shows the full size of the volume.
But I have to resize that partition, to be able to use the full volume.

maxnet, 2021-07-16:
>But I have to resize that partition, to be able to use the full volume.

How much does the size of the partition differ from the full size?

And I repeat my question: on which platform are you running Imager?

Reason I am asking is that the method used to re-partition differs by platform. 
On Windows we outsource it to "diskpart"
On Mac OS X we call "diskutil eraseDisk FAT32 SDCARD MBRFormat /dev/name-of-device".
When you run Imager on Linux as normal user partitioning is outsourced to udisks2
When Imager is run on Linux as root user partitioning is done by sfdisk

So to figure out where things go wrong, it would be handy to know which of the 4 programs is not behaving as expected...

tdewey-rpi, 2024-02-06:
Closing as stale, no replies since '21.

closed 2024-02-06 by tdewey-rpi