raspberrypi/rpi-imager #250: [OS request] Apertis

a-wai, 2021-07-16:
Hi,

I'm one of the maintainers of [Apertis](https://www.apertis.org), a Debian derivative tailored for embedded systems, with a particular focus on the automotive industry. We distribute images for 64-bit Raspberry Pi boards (Pi 3/4), and as such, we'd be pleased if you could include Apertis in the rpi-imager OS list. We have json file ready and updated for each new release, here is the OS entry we ask you to add:

```
{
    "name": "Apertis",
    "description": "Debian-based distribution aimed at embedded systems",
    "icon": "https://images.apertis.org/rpi/apertis-logo.png",
    "subitems_url": "https://images.apertis.org/rpi/apertis-oslist.json"
}
```

Thanks

a-wai, 2021-08-12:
Ping?

ghollingworth, 2021-08-12:
Do you want to add a website URL for the top level?  See the top-level .json file for an example...

ghollingworth, 2021-08-12:
I'd suggest having a page for each of the images (or a single page which you direct them all to) which describes the difference between the various options, since there are so many...

a-wai, 2021-08-13:
That sounds like a good idea, thanks for the suggestion!

For the top-level we could make it that way:
```
{
    "name": "Apertis",
    "description": "Debian-based distribution aimed at embedded systems",
    "icon": "https://images.apertis.org/rpi/apertis-logo.png",
    "subitems_url": "https://images.apertis.org/rpi/apertis-oslist.json",
    "website": "https://images.apertis.org/"
}
```

We'll add a link to https://www.apertis.org/policies/images/ (which explains the image types we provide) to each image entry on our side.

a-wai, 2021-08-13:
See https://gitlab.apertis.org/infrastructure/apertis-infrastructure/-/merge_requests/79 and https://gitlab.apertis.org/infrastructure/apertis-image-recipes/-/merge_requests/414 for our corresponding internal MRs.

ghollingworth, 2021-08-16:
Looks good, I would suggest though that the top-level link needs to provide information about what Apertis is, I think https://www.apertis.org/ gives this better than images.apertis.org

Agree the https://www.apertis.org/policies/images/ explains (in detail) what each thing means and is suitable for the links themselves...

Gordon

andrunko, 2021-08-16:
> Looks good, I would suggest though that the top-level link needs to provide information about what Apertis is, I think https://www.apertis.org/ gives this better than images.apertis.org
>
> Agree the [...]

Thanks @ghollingworth, let's use https://www.apertis.org as top-level then - @a-wai is on vacations atm and I will be following-up here if other changes are needed, please let us know.

Fwiw, the internal MRs above already landed and we should be good to go. For simplicity, here is the updated top-level:
```
{
    "name": "Apertis",
    "description": "Debian-based distribution aimed at embedded systems",
    "icon": "https://images.apertis.org/rpi/apertis-logo.png",
    "subitems_url": "https://images.apertis.org/rpi/apertis-oslist.json",
    "website": "https://www.apertis.org/"
}
```

ghollingworth, 2021-08-25:
Note: not ignoring this, the guy who normally tests / integrates the updates is on vacation.

andrunko, 2021-08-25:
> Note: not ignoring this, the guy who normally tests / integrates the updates is on vacation.

Thanks for the update, please let us know if there is anything else needed on our side.

a-wai, 2021-11-08:
@ghollingworth any news on that, please?

XECDesign, 2021-11-15:
Is this more of a general purpose OS or an "other specific purpose" OS?

ghollingworth, 2021-11-17:
I think this is just a general purpose OS...

XECDesign, 2021-11-17:
Thanks. It's in the list now.

andrunko, 2021-11-24:
Thanks all, I guess we can close this one now?

closed 2021-11-24 by a-wai

jyangchisyan, 2022-01-03:
Hi @a-wai , I found the os_list img url use ***.img in the apertis-oslist.json but the release page only found tar files. Please fix related json files or add img files.

a-wai, 2022-01-06:
@jyangchisyan thanks for noticing, this is fixed and pending review on our end, should land really soon.

ghollingworth, 2022-02-02:
Could one of the Apertis developers email me gordon@ this organisation dot com

Just so I've got a contact name

macmpi, 2024-09-28:
@a-wai would you mind adding devices filters within https://images.apertis.org/rpi/apertis-oslist.json & related subsections, like `"devices": [ "pi4-64bit", "pi3-64bit"] ` or whatever is relevant (see samples [here](https://downloads.raspberrypi.org/os_list_imagingutility_v4.json))?
Thanks!