raspberrypi/rpi-imager #251: Error formatting (through udisks2)

jpmvferreira, 2021-07-18:
Hi there, I've recently had issues with my Raspberry Pi 4 B, where both the red and green light stay always on. I've tested with two different SD cards and two different OS's (Manjaro ARM and Ubuntu Server) both headless.

Now I've been recommended to try flashing a recovery image using the `rpi-imager`, which I installed on my laptop that is running Manjaro from the [AUR](https://aur.archlinux.org/packages/rpi-imager/), which seems to be up to date to the latest version.

In the GUI I picked the following options: Misc Utility Images > Bootloader (I assume that this is the right image to try to recover a RPI 4). After that I selected my SD card, which appears to have no error associated with it and was able to flash it with `balenaetcher` with no issues raised, and after a few seconds I get the following error:
```
Error formatting (through udisks2)
```

The only post I've seen with this exact same issue is one found one [reddit](https://www.reddit.com/r/raspberry_pi/comments/gmjsve/issues_with_setting_up_sd_card/), which was deleted but a backup of the question is available on [this forum](https://www.videogamex.com/forums/showthread.php?540431-Issues-with-setting-up-SD-card), and it turns out it was an hardware issue, however, this SD card seems to be working fine and the SD card to USB adapter is brand new.

maxnet, 2021-07-18:
You did install dosfstools (pacman -S dosfstools)?
As the AUR seems to list it as an optional dependency, but for the recovery image it is not so optional.

Does it work any better if you run it with sudo rpi-imager?
In that case it will call the programs to format as FAT32 directly, instead of outsourcing it to udisks2.
(Although you may run into another issue with that: https://github.com/raspberrypi/rpi-imager/issues/237 )

jpmvferreira, 2021-07-19:
> You did install dosfstools (pacman -S dosfstools)?
> As the AUR seems to list it as an optional dependency, but for writing the recovery image it is not so optional.

Yes it's installed on my system.

> Does it work any better if you run it as root with "sudo rpi-imager"?

Actually it did, interesting. I formatted it, successfully, and then installed the RPI bootloader for SD card, which was also successfully installed. 

Any hints on how to solve this for non root users using `udisks2`?

maxnet, 2021-07-19:
>Any hints on how to solve this for non root users using udisks2?

Check if "journalctl -u udisks2" shows any errors.
Can also monitor jobs by leaving "udisksctl monitor" running on a terminal.

Would normally show something like this on job start:

```
13:31:49.223: Added /org/freedesktop/UDisks2/jobs/1
  org.freedesktop.UDisks2.Job:
    Bytes:              0
    Cancelable:         true
    ExpectedEndTime:    0
    Objects:            ['/org/freedesktop/UDisks2/block_devices/sdc']
    Operation:          partition-create
    Progress:           0.0
    ProgressValid:      false
    Rate:               0
    StartTime:          1626694309222186
    StartedByUID:       1000

```

Then some output about the disk properties changing, and at the end on success:

```
13:31:49.873: /org/freedesktop/UDisks2/jobs/1: org.freedesktop.UDisks2.Job::Completed (true, '')
13:31:49.874: Removed /org/freedesktop/UDisks2/jobs/1
```

Not sure how verbose it is on errors though...

jpmvferreira, 2021-07-19:
Here's the output of `udisksctl -monitor` when trying to erase my SD card in `rpi-imager` running as a normal user:
```
$ udisksctl monitor
Monitoring the udisks daemon. Press Ctrl+C to exit.
14:04:51.517: The udisks-daemon is running (name-owner :1.29).
14:04:57.733: Added /org/freedesktop/UDisks2/jobs/4
  org.freedesktop.UDisks2.Job:
    Bytes:              0
    Cancelable:         true
    ExpectedEndTime:    0
    Objects:            /org/freedesktop/UDisks2/block_devices/sdc1
    Operation:          filesystem-unmount
    Progress:           0.0
    ProgressValid:      false
    Rate:               0
    StartTime:          1626699897728462
    StartedByUID:       0
14:04:57.735: /org/freedesktop/UDisks2/block_devices/sdc1: org.freedesktop.UDisks2.Filesystem: Properties Changed
  MountPoints:          
14:04:57.738: /org/freedesktop/UDisks2/jobs/4: org.freedesktop.UDisks2.Job::Completed (true, '')
14:04:57.738: Removed /org/freedesktop/UDisks2/jobs/4
14:04:57.770: /org/freedesktop/UDisks2/block_devices/sdc1: org.freedesktop.UDisks2.Block: Properties Changed
  UserspaceMountOptions:        
14:04:57.848: Removed /org/freedesktop/UDisks2/block_devices/sdc1
14:04:57.899: /org/freedesktop/UDisks2/block_devices/sdc: Removed interface org.freedesktop.UDisks2.PartitionTable
14:04:57.982: /org/freedesktop/UDisks2/block_devices/sdc: Added interface org.freedesktop.UDisks2.PartitionTable
  Partitions:           
  Type:                 dos
14:04:58.061: Added /org/freedesktop/UDisks2/jobs/5
  org.freedesktop.UDisks2.Job:
    Bytes:              0
    Cancelable:         true
    ExpectedEndTime:    0
    Objects:            /org/freedesktop/UDisks2/block_devices/sdc
    Operation:          partition-create
    Progress:           0.0
    ProgressValid:      false
    Rate:               0
    StartTime:          1626699898055542
    StartedByUID:       1000
14:04:58.173: Added /org/freedesktop/UDisks2/block_devices/sdc1
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/sdc1
    DeviceNumber:               2081
    Drive:                      '/org/freedesktop/UDisks2/drives/Generic_MassStorageClass_000000001536_1'
    HintAuto:                   true
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 false
    Id:                         by-uuid-82B9-F428
    IdLabel:                    
    IdType:                     vfat
    IdUUID:                     82B9-F428
    IdUsage:                    filesystem
    IdVersion:                  FAT32
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/sdc1
    ReadOnly:                   false
    Size:                       64017661952
    Symlinks:                   /dev/disk/by-id/usb-Generic_MassStorageClass_000000001536-0:1-part1
                                /dev/disk/by-partuuid/59949d17-01
                                /dev/disk/by-path/pci-0000:00:14.0-usb-0:1:1.0-scsi-0:0:0:1-part1
                                /dev/disk/by-uuid/82B9-F428
    UserspaceMountOptions:      
[... 100 lines cut ...]
                        /dev/disk/by-uuid/87BC-6479
14:05:02.785: /org/freedesktop/UDisks2/block_devices/sdc1: Added interface org.freedesktop.UDisks2.Filesystem
  MountPoints:          
  Size:                 0
14:05:02.848: Added /org/freedesktop/UDisks2/jobs/8
  org.freedesktop.UDisks2.Job:
    Bytes:              0
    Cancelable:         true
    ExpectedEndTime:    0
    Objects:            /org/freedesktop/UDisks2/block_devices/sdc1
    Operation:          filesystem-mount
    Progress:           0.0
    ProgressValid:      false
    Rate:               0
    StartTime:          1626699902844169
    StartedByUID:       0
14:05:02.854: /org/freedesktop/UDisks2/jobs/8: org.freedesktop.UDisks2.Job::Completed (true, '')
14:05:02.856: Removed /org/freedesktop/UDisks2/jobs/8
14:05:02.857: /org/freedesktop/UDisks2/block_devices/sdc1: org.freedesktop.UDisks2.Filesystem: Properties Changed
  MountPoints:          /run/media/undercover/87BC-6479
14:05:02.889: /org/freedesktop/UDisks2/block_devices/sdc1: org.freedesktop.UDisks2.Block: Properties Changed
  UserspaceMountOptions:        uhelper=udisks2
```

Not quite sure what went wrong here, as it seems that the job is completed successfully, but there's a lot of output that I'm not quite sure what it means.

jpmvferreira, 2021-07-19:
Here's the output of running `rpi-imager` as a regular user in the terminal and trying to erase my SD card:
```
$ rpi-imager
QObject::setParent: Cannot set parent, new parent is in a different thread
Repartitioning drive
Adding partition
New partition: "/org/freedesktop/UDisks2/block_devices/sdc1"
Formatting drive as FAT32
Mounting partition
Mounting partition
Mounting partition
Mounting partition
Mounting partition
Mounting partition
Mounting partition
Mounting partition
Mounting partition
Mounting partition
Failed to mount new file system.
```

It does seem to be an issue with mounting the SD card, but it does seem to be mounted, and after reinserting the USB adapter that I use to read the SD card it successfully mounts the new file system.

For comparison here's the output of running `sudo rpi-imager` that doesn't udisks2:
```
$ sudo rpi-imager
[sudo] password for undercover: 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QObject::setParent: Cannot set parent, new parent is in a different thread
sfdisk: "Checking that no-one is using this disk right now ... OK\n\nDisk /dev/sdc: 59.63 GiB, 64021856256 bytes, 125042688 sectors\nDisk model: MassStorageClass\nUnits: sectors of 1 * 512 = 512 bytes\nSector size (logical/physical): 512 bytes / 512 bytes\nI/O size (minimum/optimal): 512 bytes / 512 bytes\nDisklabel type: dos\nDisk identifier: 0x59949d17\n\nOld situation:\n\nDevice     Boot Start       End   Sectors  Size Id Type\n/dev/sdc1        8192 125042687 125034496 59.6G  e W95 FAT16 (LBA)\n\n>>> Created a new DOS disklabel with disk identifier 0xd7a4ad20.\n/dev/sdc1: Created a new partition 1 of type 'W95 FAT16 (LBA)' and of size 59.6 GiB.\nPartition #1 contains a vfat signature.\n/dev/sdc2: Ignoring partition.\n/dev/sdc3: Ignoring partition.\n/dev/sdc4: Ignoring partition.\n/dev/sdc5: Done.\n\nNew situation:\nDisklabel type: dos\nDisk identifier: 0xd7a4ad20\n\nDevice     Boot Start       End   Sectors  Size Id Type\n/dev/sdc1        8192 125042687 125034496 59.6G  e W95 FAT16 (LBA)\n\nThe partition table has been altered.\nCalling ioctl() to re-read partition table.\nSyncing disks.\n"
mkfs.fat: "mkfs.fat 4.2 (2021-01-31)\n"
```

maxnet, 2021-07-19:
Perhaps something else on the system auto-mounting the file system, and subsequent mount attempts failing because already mounted.
Guess we could add some code to check for that race condition.

jpmvferreira, 2021-07-19:
Yes, but it seems that is something that is not specific to the `rpi-imager`, so feel free to close this issue, unless you want to keep it open as a feature request to add that feature of race condition.

Thanks!

2-bithacker, 2021-09-20:
What I've found to fix that glitch is a total reboot. I reboot and it works when I try again. This is a crude way to o it but it works for me.

AloisJanicek, 2021-09-25:
Exactly like @maxnet  pointed out, I had enabled automatic mounting on my KDE Plasma desktop. Once I disabled it I was able to write the boot loader changes successfully on the SD card.
Maybe adding this issue into some kind of FAQs and link them from rpi-imager's UI through little "?" help icon would be not a bad idea.

tdewey-rpi, 2024-02-06:
Closing as:

1. Stale
2. Not an issue that can be fixed by changes to rpi-imager

closed 2024-02-06 by tdewey-rpi