raspberrypi/rpi-imager #252: How to create a password for the user for firstrun.sh from the commandline?

laszewsk, 2021-07-19:
How is the password for firstrun.sh created when it is entered via the GUI?

We like to change it through a command line tool and not the GUI, Has anyone a shell script that does create such a password? I am looking for a command that does the following

encrypt_passwd.sh password    # ???? how do we do this? 

prints out the password we need to place in firstrun.sh. Python prg would be best, but sh script would also work

lurch, 2021-07-19:
> How is the password for firstrun.sh created when it is entered via the GUI?

https://github.com/raspberrypi/rpi-imager/blob/qml/OptionsPopup.qml#L519

https://github.com/raspberrypi/rpi-imager/blob/qml/imagewriter.cpp#L955

> Python prg would be best, but sh script would also work

https://passlib.readthedocs.io/en/stable/lib/passlib.hash.sha256_crypt.html

maxnet, 2021-07-19:
If you intend to run that Python script on a Linux flavour, you can probably also just use the standard crypt() function.

https://docs.python.org/3/library/crypt.html

laszewsk, 2021-07-20:
thanks for pointing out the c code location. We converted and we can now set the password in firstrun.sh with python. It works!

closed 2021-07-20 by laszewsk