raspberrypi/rpi-imager #253: Could we get a portable version too?

shukriadams, 2021-07-25:
Feature request: would it be possible to get the imager in a portable format so we don't have to install it? Also, thanks for making this.

tdewey-rpi, 2024-02-06:
I don't have plans to implement this in the near term, as it would require statically linking Qt.

It's possible in the future we may look at this, but I can't see the entry criteria for this feature happening any time soon.

As such, closing for tracking.

closed 2024-02-06 by tdewey-rpi

maxnet, 2024-05-09:
>I don't have plans to implement this in the near term, as it would require statically linking Qt.

"Portable" does not necessarily mean "single file"

Technically, on Windows, you can already copy the files from "\program files (x86)\raspberry pi imager" to USB stick and run those on another computer. It should not care about the location of the files. Just expects the Qt .dll's to live in the same folder as rpi-imager.exe

Only thing that is not portable are the settings.
If that do is desired, you could add some code to the beginning of main() that detects whether it is the portable version, and if the folder that rpi-imager.exe is in is writable by the user, and if so configure Qt to store settings in an .ini file instead of registry.
Recall you can do so with code among the lines of:

```
QSettings::setDefaultFormat(QSettings::IniFormat);
QSettings::setPath(QSettings::IniFormat, QSettings::UserScope, QCoreApplication::applicationDirPath() );
```

(untested)