raspberrypi/rpi-imager #254: Add an option to define partition sizes before installing on the expert menu.

gscorrea, 2021-07-26:
Would like to define the size of patrtitions before installation, so we can resize at least the boot partition.
With new SD cards we can have /boot with 5G and not have issues with upgrading OS.

lurch, 2021-07-26:
To do that, Rpi Imager would need to be able to also resize the filesystems inside the partitions, and on Windows it's not possible to resize Linux filesystems.

gscorrea, 2021-07-27:
The issue is that the /boot partition is too small and do not allow to add more kernels.

tdewey-rpi, 2024-02-06:
Marking as 'Won't fix'.

I don't see new or inexperienced users getting benefit from this feature, nor do I see clearly how to expose it in a manner where an inexperienced user doesn't do the 'wrong' thing.

closed 2024-02-06 by tdewey-rpi