raspberrypi/rpi-imager #258: Capturing rpi-imager.exe console output in python

sadilekivan, 2021-08-04:
After some trouble I found out that this great tool can be used in a CLI manner with its option. I would like to autoselect my sdcard with my python script. To do that I need to get the output of the tool, but unlike most of other programs I cant get it to work.

Heres a little test example, the output of the command should be supressed, captured into stdout and then printed between the Start and End word.

```py
import subprocess

stdout = subprocess.run(['echo', 'executing the echo command'], capture_output=True, text=True).stdout
print(f'Start\n{stdout}End')

stdout = subprocess.run(['c:/Program Files (x86)/Raspberry Pi Imager/rpi-imager.exe', '--cli', '2021-05-07-raspios-buster-armhf-lite.img', 'list'], capture_output=True, text=True).stdout
print(f'Start\n{stdout}End')
```

Output:

```
Start
executing the echo command
End
[INFO] Enumerating SCSI
[INFO] Enumerating USBSTOR
[INFO] Enumerating SCSI
Destination drive is not in list of removable volumes. Choose one of the following:

\\.\PhysicalDrive1 (asmedia ASM1153 SCSI Disk Device)
\\.\PhysicalDrive2 (Generic STORAGE DEVICE USB Device)

Or use --enable-writing-system-drives to overrule.
Start
End
```

How does the rpi-imager send data to the terminal? It just doesnt seem to be there with stdout 😞.

maxnet, 2021-08-04:
>How does the rpi-imager send data to the terminal?

Problem may be that rpi-imager is compiled as a GUI program and not as a CLI program, and Windows only allows output redirection for CLI programs.

May work better if you recompile from source with WIN32 removed here: https://github.com/raspberrypi/rpi-imager/blob/qml/CMakeLists.txt#L86
Have not tried it though.

See the CMake manual for what that does:
https://cmake.org/cmake/help/latest/command/add_executable.html
https://cmake.org/cmake/help/latest/prop_tgt/WIN32_EXECUTABLE.html#prop_tgt:WIN32_EXECUTABLE

marcomoerz, 2022-11-08:
Is there a new development for capturing the output of the application with python?

maxnet, 2022-11-08:
>Is there a new development for capturing the output of the application with python?

Should work fine under Linux.
No new developments for Windows.

Also not sure if there is much demand for it.
Keep in mind that if you have a lot of devices to install, there may be other better options (such as cmprovision) for it.
But that depends on your exact use-case.

tdewey-rpi, 2024-02-06:
Closing as not-viable for Windows. Alternative tools, as mentioned by @maxnet, might be a better fit for your use case.