raspberrypi/rpi-imager #259: [OS Update Request] New JSON snippet for OctoPi

foosel, 2021-08-05:
Hi there! In #162, OctoPi was added to the OS list - thanks for that, that's been a huge improvement for the OctoPrint community! 

Today I'd like to request a change of the `os-list` JSON snippet to this:

```json
{
    "name": "OctoPi",
    "description": "A Raspberry Pi distribution for 3D printers. Ships OctoPrint out-of-the-box.",
    "website": "https://octopi.octoprint.org",
    "icon": "https://octopi.octoprint.org/rpi-imager.png",
    "subitems_url": "https://octopi.octoprint.org/rpi-imager.json"
}
```

The goal is to change the JSON URL to a new canonical place and also add a website as recently suggested by @ghollingworth.

Pinging @guysoft so he can confirm.

Screenshots:

[image]

[image]

guysoft, 2021-08-05:
I confirm :)

XECDesign, 2021-08-06:
Updated

foosel, 2021-08-06:
Thank you, just confirmed!

closed 2021-08-06 by foosel