raspberrypi/rpi-imager #267: Add option to specify the IP Address using cmdLine.txt files IP=x.x.x.x  [Feature Request]

PJTewkesbury, 2021-08-18:
It would be really useful to add an option to the Advanced settings to enable the user to define the IP address to use for Raspberry PI using the cmdline.txt file's IP=x.x.x.x option.

Thanks

lurch, 2021-08-18:
My uninformed guess would be that 90% of users just use whatever IP address their DHCP server (e.g. in their internet router) allocates them, so adding an option to manually specify the IP address might confuse them? :man_shrugging:

ghollingworth, 2021-08-23:
Also, the ip= option will only force a fixed IP address for the boot phase of the process, once the Raspberry Pi has booted this option is ignored and dhcpcd handles assignment of the IP address

closed 2021-08-23 by ghollingworth