raspberrypi/rpi-imager #268: Lzma library error: Corrupted input data

bredmor, 2021-08-20:
Version: `1.6.2` (Ubuntu SNAP)

I received this error while trying to write the Ubuntu Server 64bit image from the `Choose OS` menu to an SD card, several days after successfully writing to multiple SD cards.

I would guess that the issue is that the cached image for my chosen OS is corrupted, however the application does not give me any option to delete the image, re-download the image, nor does it tell me *where* the images are cached. I could not find anything apparent in `~/.cache` or `/tmp`.

I would suggest an option to delete the cached image on any kind of extraction failure.

In the mean time, I would appreciate information on where the image files are stored so that I may delete them if possible. I did peruse the source for a few minutes and found the functions for writing to the cache, but unfortunately was unable to immediately find where the location is specified.

maxnet, 2021-08-29:
>I would suggest an option to delete the cached image on any kind of extraction failure.

We already discard the cache file at the end of the download if the advertised SHA256 checksum does not match.
So this should not be happening, the file gets corrupted later after saving. E.g. if your storage is actually failing.

>Version: 1.6.2 (Ubuntu SNAP)

Suggest you use the .deb we provided instead.
Or ask here where the snap sandbox saves it files: https://github.com/popey/imager-snap

(The .deb saves it at ~/.cache/Raspberry Pi/Imager/lastdownload.cache . But no idea where sandboxed versions like snap store their stuff).

closed 2021-11-29 by bredmor

wzqvip, 2023-12-05:
That helps a lot. I used 7zip to verify and found error at the end of the file.