raspberrypi/rpi-imager #269: [Feature Request] Show password checkbox for ssh

SaXeee, 2021-08-25:
Would be great to have in "advanced options" a "show password" checkbox for the ssh section like the one already available for the WiFi.

p.s. thank you for your awesome job.

snapdeus, 2021-09-22:
My recommendation/request would be to require inputting the password twice and making sure they match instead of a show password option.  Either one would work.

reukiodo, 2021-12-02:
There is already a 'Show password' checkbox for the WiFi password, so it would seem to follow good UX to do the same way for the SSH password.

maxnet, 2021-12-02:
They are handled a bit different under the hood though.
Wifi password is saved as-is, so can indeed be shown. Even if it was saved in a previous session.

SSH password is stored hashed ( crypt() ).
(Can also see that the number of asterikses do not match your password if it was saved)

cillian64, 2023-10-19:
I think it's more important that we store the SSH password in a hashed format, I don't like the security implications of storing the password in cleartext