raspberrypi/rpi-imager #271: How to change language

Wowooco, 2021-08-29:
I noticed that there is a folder "`i18n`" in this warehouse. In fact, I am not used to using English, but I am used to my mother tongue, so I would like to ask how to change the language, because I did not find the button

maxnet, 2021-08-29:
It will use the language of your operating system.

Wowooco, 2021-08-29:
[image]
What might be the problem? I set it up as you said, but it didn’t take effect!

Wowooco, 2021-08-29:
> 它将使用您操作系统的语言。
[image]
What might be the problem? I set it up as you said, but it didn’t take effect

maxnet, 2021-08-29:
May need to restart after changing it.

Also note that we only currently have a translation for zh-CN, which I think is mainland China, simplified characters.
If you have some different variant, like zh-TW or zh-HK it will not work.

Wowooco, 2021-08-29:
> May need to restart after changing it.
> 
> Also note that we only currently have a translation for zh-CN, which I think is mainland China, simplified characters.
> If you have some different variant, like zh-TW or zh-HK it will not work.

I'm sure I set zh_CN, Windows restarted, and other versions of rpi-imager were installed, but it didn't take effect.
System: Windows 10 Pro 21H1
Rpi-imager: 1.6.2

maxnet, 2021-08-29:
Hmm, seems to be indeed a bug there.

Judging by the output of "powershell" "Get-WinUserLanguageList" it seems that Windows uses "zh-Hans-CN".
(On Linux and MacOS X it does seem to be zh_CN, so if the person that contributed the Chinese translations was using that, he may not have noticed).

Wonder if we should just change rpi-imager_zh_cn.ts to rpi-imager_zh.ts to have it match everything starting with "zh".

Wowooco, 2021-08-29:
> Hmm, seems to be indeed a bug there.
> 
> Judging by the output of "powershell" "Get-WinUserLanguageList" it seems that Windows uses "zh-Hans-CN".
> (On Linux and MacOS X it does seem to be zh_CN, so if the person that contributed the Chinese translations was using that, he or she may not have noticed).
> 
> Wonder if we should just change rpi-imager_zh_cn.ts to rpi-imager_zh.ts to have it match everything starting with "zh".

What should I do, because I don’t know how to compile software by myself

maxnet, 2021-08-29:
We will take care of it in a future version.
Unfortunately do not have a quick solution right now. (and am about to catch a plane, so cannot build it for you right now)

commit 6a1744e2 (2021-11-18): Windows: fix Chinese language

Closes #271
Closes #272
closed 2021-11-18 by maxnet