raspberrypi/rpi-imager #274: [OS request] Mainsail OS

meteyou, 2021-09-07:
Hi,

I'm the maintainer of [mainsail](https://github.com/meteyou/mainsail) and we have also a prebuild raspberry pi image with [klipper](https://github.com/kevinOConnor/klipper/), [moonraker](https://github.com/arksine/moonraker) and [mainsail](https://github.com/meteyou/mainsail). This OS is called "Mainsail OS".

I have created the JSON entry:
```JSON
{
          "name": "Mainsail OS",
          "description": "Klipper Firmware & Moonraker API and Mainsail UI - ready to print!",
          "website": "https://os.mainsail.xyz",
          "icon": "https://os.mainsail.xyz/rpi-imager.png",
          "subitems_url": "https://os.mainsail.xyz/rpi-imager.json"
}
```
Thank you

lurch, 2021-09-08:
> `"description": "Klipper Firmware & Moonraker API and Mainsail UI - ready to print!"`

Seems to assume that people are already aware of what Klipper, Moonraker & Mainsail are?? I've certainly never heard of any of them before. Perhaps a more generic description would be more appropriate? :shrug:

BastelKlug, 2021-09-13:
@lurch Mainsail (frontend) with Moonraker (backend) is like Octoprint and Klipper is the firmware. I think @meteyou just have to add the word 3D printing and everyone who has a 3D printer and will use Mainsail with Klipper will know what it is. But in general this description is appropriate.

meteyou, 2021-09-14:
sry for the delay... we are currently working on a better description that can cope with the limited space.

meteyou, 2021-10-03:
so... here is a "new" JSON entry:

```json
{
  "name": "Mainsail OS",
  "description": "Mainsail 3D printer interface - including Klipper firmware & Moonraker API. ",
  "website": "https://os.mainsail.xyz",
  "icon": "https://os.mainsail.xyz/rpi-imager.png",
  "subitems_url": "https://os.mainsail.xyz/rpi-imager.json"
}
```

Even if you don't know Mainsail and/or Klipper, I think it makes sense to name them in the description. Klipper is the rising star among 3D printer firmware and should be mentioned because it is the only firmware that can communicate with Mainsail.
Marlin, RRF and Smoothie are not supported.

meteyou, 2021-10-14:
@lurch would this description fit or do you have any change requests?

ghollingworth, 2022-02-02:
Could you please email me on gordon @ this organisation dot com

Thanks

ghollingworth, 2022-02-02:
Added

closed 2022-02-02 by ghollingworth