raspberrypi/rpi-imager #278: Rpi-Imager has an error downloading the os list

2-bithacker, 2021-09-20:
I try to do an install of ubuntu on a newly formatted 64 gb sd card and the install takes about 8 or 9 hours and as a failed verification or takes about 10 minutes and has an error downloading the os.

lurch, 2021-09-20:
> I try to do an install of ubuntu on a newly formatted 64 gb sd card and the install takes about 8 or 9 hours

There's no way it should take that long I think! Sounds like either your SD-card or your card-reader has problems? Do you have a different SD-card and/or a different card-reader you can use for testing?

> has an error downloading the os

Maybe there's also a problem with the stability of your internet connection? :man_shrugging:

2-bithacker, 2021-09-20:
My internet connection is fine. I browse the web, and download stuff all the time. Imager says it is something about the host site not being closed properly, and I don't know how that can be.

maxnet, 2021-09-20:
>I try to do an install of ubuntu

Did you try any of the other operating systems?
Just to rule out that there is an issue with your local Ubuntu mirror.

2-bithacker, 2021-09-20:
Yes. Raspberry pi os was the one that took 15 minutes to fail to download, and Manjaro arm linux and Ubuntu are the ones that took 10 hours to fail to verify.

lurch, 2021-09-20:
> Raspberry pi os was the one that took 15 minutes to fail to download

What happens if you try downloading it manually from https://www.raspberrypi.org/software/operating-systems/ and then flashing the downloaded IMG file with RPi Imager?

2-bithacker, 2021-09-20:
I tried that with ubuntu and raspbian, and it takes 10 hours to fail to verify.

lurch, 2021-09-20:
IMHO that points to a problem with either your SD-card or card-reader, as I mentioned earlier. You could try using [Etcher](https://www.balena.io/etcher/), but I presume that'll probably be ridiculously slow for you as well.

lurch, 2021-09-20:
Ah, you never mentioned that you were using a Raspberry Pi; unfortunately Etcher [don't provide ARM binaries](https://github.com/balena-io/etcher/issues/2870) yet, so apologies for sending you on a wild goose chase.

> but the cpu frequency is 2.6 ghz, and the gpu frequency is 850 mhz

I think those are well above the stock frequencies, so maybe it's worth removing all overclocking options _in case_ that makes any difference? :shrug:  Also, do you get the same performance results when using both the USB2 and USB3 ports on the Raspberry Pi 4B?

You could also try [these instructions](https://www.raspberrypi.org/documentation/computers/getting-started.html#installing-images-on-linux) if you're comfortable with the command line?

2-bithacker, 2021-09-21:
Okay thanks for telling me about that. I should have told about the raspberry pi part the first time. As for the usb, the adapter is for usb 2.0, so I use the 3.0 ports for the keyboard and mouse because I do not have any 3.0 devices, and I use the 2.0 ports for hard drives and stuff like that. I will try the undo the overclock, and see if that works.

2-bithacker, 2021-09-21:
I undid the overclock, and it still does the same thing. I'll try those instructions you referred me to.

lurch, 2021-09-21:
I just did a quick test myself, and running RPi-Imager on a Raspberry Pi 400, with a (good quality) USB card-reader plugged into one of the USB2.0 ports, it took just 6 minutes to write and verify Raspberry Pi OS onto a (good quality) 32 GB microSD card, which is much more in line with what I'd expect. So there's definitely **some part** of your system that isn't working correctly, and the only way to figure out what will be trial and error. My first suggestion would be to try a different card-reader (see e.g. https://github.com/raspberrypi/rpi-imager/issues/212#issuecomment-822841336 )

2-bithacker, 2021-09-21:
I have tried 4 different models of sd card readers that came with some drones that I have gotten over the years. Two are really cheap-looking, and they do not work. Two other higher quality looking ones do work. I have been thinking  the sd card might have been formatted improperly. I tried a 1 tetrabyte sd card and a 64 gigabyte card. I looked at a tutorial for formatting a sd card larger than 32 gigabytes for raspberry pi said to do FAT32. Is that right? The reason I thought that it may be the format is wrong because I have installed other operating systems such as raspberry pi os on a 16 gigabyte card and ubuntu lite on a 4 gigabyte card. I also thought that raspberry pi imager on my system has been corrupted somehow so I reinstalled it. It still has that problem. I tried that commandline tutorial you referred me to and it doesn't seem to work. Thank you for going into such efforts to help.

lurch, 2021-09-21:
> I tried a 1 tetrabyte sd card

From a quick look on Amazon.co.uk, legitimate 1TB microSD cards are around the £200 mark, so I suspect you've probably bought a "fake" microSD card, in which case it's not surprising that it's not working properly.

2-bithacker, 2021-09-21:
Strange, the reviews on it said the 2 terabyte sd cards were great, according to about 5 people. And yes, I paid nothing close to $250. More like $10. These things are all over ebay, So, I'll look in to it. Here are some reviews that refer to this subject on the ebay seler, https://www.ebay.com/usr/andy5081 ,that I got it of of:
"I recognized the simple packaging as the only brand that works right out of the package. It is really 2TB and that's no misprint!"
"bought 3 1t sd cards and not a single one works... very disappointing."
"GREAT 2TB Micro card, Works Great, GREAT price, fast shipping and work the first time !!!"
"was not class 10" subject:(about the 1 tetrabyte)
What do you think? I think the 1 tetrabyte ones are bad, and the 2 tetrabyte ones that this ebay seller sells are good.

lurch, 2021-09-21:
> ... ebay ...

There's your problem! :stuck_out_tongue_winking_eye: :rofl:

> I paid nothing close to $250. More like $10.

If something seems to good to be true, it normally is.

> the 2 tetrabyte ones

AFAIK 2TB microSD cards don't exist (and a quick google search seems to confirm that), so they're clearly all fakes. Which you can read more about at e.g. https://sosfakeflash.wordpress.com/

2-bithacker, 2021-09-21:
Okay thanks.

closed 2021-09-21 by 2-bithacker