raspberrypi/rpi-imager #280: Imager not opening

lipflip, 2021-09-23:
Just installed imager 1.6.2, the imager won't launch and I don't see an icon in taskbar after launching.

I only have one screen attached.
I'm using Windows 10

See:

#71
#220 

Strange thing is that I don't have those registery entries mentioned in #220

HKEY_CURRENT_USER\SOFTWARE\Raspberry Pi\Imager

lurch, 2021-09-23:
Do you get any output text if you run `C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe` in a command prompt window?

lipflip, 2021-09-23:
> C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe

I don't get any output text inside the command prompt window when executing through CMD, but I get a Windows popup to allow the application (I think for admin rights), then I click on Yes, and then nothing happens.

lurch, 2021-09-23:
Just out of curiosity, could you try the same thing again but launch the command-prompt itself as an admin user first? (which I think you can do by right-clicking on it's icon?)

maxnet, 2021-09-23:
The other issues you link are unrelated.
When the position of the Imager window is outside the connected screen, it still appears on taskbar, and is visible in alt+tab, etc.

Windows did not indicate the application crashed?
(That can also be viewed in the event log.
[Windows key] + R -> "eventvwr"
If a crash occured it should have have an entry similar to this:

[image]
)

Imager/Qt uses graphics acceleration, and as such depends on proper graphics drivers being installed.
Other applications that use graphics acceleration (e.g. games), do run without problems on your computer?

lipflip, 2021-09-23:
> 
> 
> Just out of curiosity, could you try the same thing again but launch the command-prompt itself as an admin user first? (which I think you can do by right-clicking on its icon?)

I tried this with admin rights, but then nothing happens.

lipflip, 2021-09-23:
> 
> 
> The other issues you link are unrelated.
> When the position of the Imager window is outside the connected screen (the other issues), it still appears on taskbar, and is visible in alt+tab, etc.
> And you indicate it not being on the taskbar.
> 
> Windows did not indicate the application crashed?
> (That can also be viewed in the event log.
> [Windows key] + R -> "eventvwr"
> If a crash occured it should have have an entry similar to this:
> 
> [image]
> )
> 
> Imager/Qt uses graphics acceleration, and as such depends on proper graphics drivers being installed.
> Other applications that use graphics acceleration (e.g. games), do run without problems on your computer?

I do have an event error in event log.

`Naam van toepassing met fout: rpi-imager.exe, versie: 1.6.2.0, tijdstempel: 0x00000000
Naam van module met fout: rpi-imager.exe, versie: 1.6.2.0, tijdstempel: 0x00000000
Uitzonderingscode: 0xc0000005
Foutmarge: 0x000fce02
Id van proces met fout: 0x7ec4
Starttijd van toepassing met fout: 0x01d7b06925d3fc97
Pad naar toepassing met fout: C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe
Pad naar module met fout: C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe
Rapport-id: 4ccb97e4-2e37-42c6-a8cc-69ec965a9adf
Volledige pakketnaam met fout: 
Relatieve toepassings-id van pakket met fout: `

That's it. (Language is in Dutch)

maxnet, 2021-09-23:
No idea why it is crashing in your case.
Could be a bug in Imager itself, but also some other library on your system that is used indirectly.

(While it says that rpi-imager.exe is the faulting module, there is no real code at the offset listed in the error report.

```
0:000> ln 0x400000+0xfce02
(00000000`004f9da0) rpi_imager!qt_resource_data+0x3062 | (00000000`0057cea0) rpi_imager!qt_resource_struct
```

Meaning the problem could also be elsewhere. Making things hard to debug)

G3077r3yC0rc0ran, 2021-09-24:
Hello Devs,

Pls help.

I also just installed rpi-imager 1.6.2. The imager won't launch and the system locks up, necessitating a hard reboot.
I tried it several times. Same result.
Then I tried to install the snapd version, but a mirror was down and the process didn't complete.
I've removed the imager, and rebooted the system.
System details are:
Raspberry Pi 4 - 4GB, Argon 1 M.2 running latest raspbian on the SSD.

Any help is gratefully received.
Thanks.

lurch, 2021-09-24:
@G3077r3yC0rc0ran I suspect that any problems with RPi Imager on Raspbian (did you actually mean Raspberry Pi OS?) are likely to be different to problems on Windows, so you should open a separate issue.

maxnet, 2021-09-24:
>the system locks up
>Raspberry Pi 4

Hangs are usually caused by:

1) Displays with bad EDID.

In that case it do should work properly if you start a terminal and enter:

```
QT_AUTO_SCREEN_SCALE_FACTOR=0 rpi-imager
```

Related to: https://github.com/raspberrypi/linux/issues/4149

2) Insufficient power supply.

3) Overclocking of any kind.

G3077r3yC0rc0ran, 2021-09-25:
Hello @lurch 
Neofetch reports:
OS: Raspbian GNU/Linux 10 (buster) armv7l

> ..., so you should open a separate issue.

OK. Will do. Thanks. :-/ N00b

Hello @maxnet 

RE: 1 Displays with bad EDID.

> pi@raspberrypi:~ $ QT_AUTO_SCREEN_SCALE_FACTOR=0 rpi-imager
> qt5ct: using qt5ct plugin
> QObject::setParent: Cannot set parent, new parent is in a different thread

Thanks a bunch!, it worked. I could see rpi-imager, and it seemed to work OK. If not, as per @lurch's suggestion, I'll start another thread. It turns out I have a crappy monitor. :-(

RE: 2 Insufficient power supply.
I have an official RP PS. So, it's probably not that.

RE: 3 Overclocking of any kind.
I've maxed it out. I'll turn off OC'ing, and see if that does it.

Thanks a bunch guys, for getting back to me, and helping me sort it out.

closed 2023-10-19 by cillian64