raspberrypi/rpi-imager #283: EEPROM Bootloader recovery needs to limit FAT32 partition size

jtbraun, 2021-10-05:
I used the imager w/ a 128 GB SD card to create a bootloader-recovery card.
When I used it in my RPI4b, I got no ACT blinking, and no video output (monitor detected no stream).

The FAT32 partition created was ~119GB large according to Windows Disk/Volume manager.

I copied the files off the SDCARD, and manually created a single 16GB FAT32 partition, and copied the files back, and rebooted the pi. That worked fine, I got a flashing ACT light and the green HDMI0 screen.

The early recovery path must have limits on the size of the FAT32 partition, or the cluster size. The RPI Imager should respect those, and create smaller FAT32 partitions on large cards.

jtbraun, 2021-10-05:
To be clear, an Ubuntu 64-bit image on this same SD Card, formatted to the "full" size boots fine w/ the bootloader EEPROM. So I agree @lurch, this is probably a limitation of the FAT32 code in the ROM that is looking for recovery.bin, or it's a limitation in recovery.bin itself (I don't have a way to tell which, I think)

lurch, 2021-10-05:
> or it's a limitation in recovery.bin itself

Ah yes, I guess that's a possibility! Perhaps this is relevant? https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md#2021-09-22---update-recoverybin-to-fix-issue-with-large-fat-partitions---stable (i.e. maybe a newer `recovery.bin` than is (currently) flashed by RPi Imager would fix the problem?)

jtbraun, 2021-10-05:
It could be. I was using the recovery.bin put down by the RPI Imager:
`SHA256(e:\recovery.bin)= 330d80aa07c3e9f17e8ce2df61bd9917e1c02d517f117dd18cce97997c8560b5`
And that's the same as the one in `rpi-eeprom/firmware/critical/recovery.bin` as of rpi-eeprom's b93a7569a3b670078917ef0c29186cd2cfc47184. That's also the "stable" version.

The beta recovery.bin is different...

timg236, 2021-10-05:
The issue is with the C0 ROM failing to load recovery.bin if the size rounded up to the nearest cluster is greater than something like 110KB.
Well probably just change the imager to I use the raw disk images and see if rpi-eeprom-update can probe the cluster size

tdewey-rpi, 2024-02-06:
Closing as stale - no further comments since '21.

timg236, 2024-02-06:
Could actually mark this as resolved because a year or two ago we modified the RPi Imager bootloader recovery images to be a pre-canned FAT image with the appropriate cluster sizes (rather than a .zip file)

closed 2024-02-06 by tdewey-rpi