raspberrypi/rpi-imager #287: [Feature Request] Set default username via advanced options menu

saltedlolly, 2021-10-13:
Currently you can set the hostname and the password, but are unable to set the default username.

maxnet, 2021-11-19:
I do think an option to specify a custom username do would be a good thing to have.
Especially taken into account that we offer more distributions than just RPI OS, and figuring out the default user (is it "pi", "ubuntu", "arch", something else?) is confusing at times when trying out a new one.
Furthermore having the username match what you have on your normal desktop computer, makes it easier to ssh to a box (can then "ssh hostname.local" instead of "ssh pi@hostname.local")

>https://www.raspberrypi.com/documentation/computers/configuration.html#changing-your-username mentions that "Please note 
>that with the current Raspberry Pi OS distribution, there are some aspects that require the pi user to be present."

Is there a list somewhere about things that do not work with pi user?
Do note that we do not use the pi user (and even delete it) on piserver deployments either, and everything still seems pretty functional....

lurch, 2021-11-19:
> Is there a list somewhere with things that do not work without pi user?

ping @spl237 and @XECDesign

XECDesign, 2021-11-19:
I don't think that statement is true anymore. If you don't forget to update the files containing the autologin user, it seems fine. I just wouldn't sign off on it saying 100% everything will work and nobody will run into any issues.

maxnet, 2021-11-19:
>I don't think that statement is true anymore.

Good.

Added some basic  customusername support.
When using the firstrun.sh method of OS customization (that is currently used for RPI OS) it will rename the Pi user to the desired username, and fix up lightdm and getty autologin.

When cloudinit customization is used (currently only by Ubuntu Server) nothing special is done on the Imager side, and I am expecting OSes to do whatever special is necessary themselves.
(If "- default" is not specified in the "users:" section, distros are expected to delete any default users they created themselves. So would argue that undo'ing any special autologin or other settings for the previous default user is also responsibility of distro)

XECDesign, 2021-11-19:
In case any of this is helpful:

```
  if [ "$MAIN_USER" != "$NEW_USER" ]; then
    mv "$OLD_HOME" "/home/$NEW_USER"
    usermod -l "$NEW_USER" -d "/home/$NEW_USER" "$MAIN_USER"
    groupmod "$MAIN_USER" -n "$NEW_USER"
    if [ -f /etc/systemd/system/getty@tty1.service.d/autologin.conf ]; then
      sed -i "s/--autologin $MAIN_USER /--autologin $NEW_USER /" /etc/systemd/system/getty@tty1.service.d/autologin.conf
    fi
    if [ -f /etc/lightdm/lightdm.conf ]; then
      sed -i "s/^\(#\|\)autologin-user=.*/autologin-user=$NEW_USER/" /etc/lightdm/lightdm.conf
    fi
    if [ -f /etc/sudoers.d/010_pi-nopasswd ]; then
     sed -i "s/^$MAIN_USER /$NEW_USER /" /etc/sudoers.d/010_pi-nopasswd
    fi
  fi
```

It's probably just the nopasswd part missing.

maxnet, 2021-11-20:
>It's probably just the nopasswd part missing.

Correct.
Fixed it.

JinShil, 2022-05-21:
Isn't this implemented now?

closed 2023-12-23 by tdewey-rpi