raspberrypi/rpi-imager #288: [Feature Request] Set Static IP addresses via advanced options menu

earth4s, 2021-10-13:
Have an option to be able to set Static IP address in the advanced option menu.

This is a useful feature in headless installs or other embedded situations. Having a pre-assigned IP address takes a lot of the guess work out what the local DHCP assigned the device or if their is not a DHCP implemented on the local network segment.

lurch, 2021-10-13:
Duplicate of #243 ?

lurch, 2021-10-14:
Haha, I found it by searching for "static" :wink: 

> Do you want to close my request or keep it separate?

That's up to @maxnet :man_shrugging:

tdewey-rpi, 2024-02-06:
Closing as a duplicate.

This feature request has been rejected elsewhere because it is unlikely to impact non-technical users, and there's not an immediately obvious way to offer this to non-technical users in a manner that is easily diagnosed and rectified. Use static DHCP leases in a DHCP environment, or a custom-rolled image for more specialised cases.

closed 2024-02-06 by tdewey-rpi