raspberrypi/rpi-imager #290: [Feature Request] Inteligent Target Device Handling

robbieb43, 2021-10-17:
I have dual boot PC with two SSDs - one for each PC OS.  I run imager under Ubuntu which is on the second SSD.  Imager offers this as first option on which to install the Pi OS.  Perhaps a more intelligent sorting of Block Devices or even an option to exclude named devices may prevent the inevitable disaster that  awaits me and other clumsy users?  

Thanks, Rob.

maxnet, 2021-10-17:
So how do you propose that we detect that the other drive -that is not being used by the currently running operating system-, is an important drive that contains important files?

maxnet, 2021-10-17:
BTW does it concern NVMe SSDs?

We used to filter those out. But removed the block ON LINUX ONLY at the request of folks wanting to provision NVMe drives for CM4 systems.
https://github.com/raspberrypi/rpi-imager/commit/906ceabaecf060e0720dd2c71e140f8347101b42

robbieb43, 2021-10-17:
On NVM, I agree there would be issues in something such as blanket blocking a single technology, though again an option to do so could work.

lurch, 2021-10-18:
If you're that worried about overwriting one of your SSDs in your main PC, I guess you could plug a USB SD card-reader into a Raspberry Pi and run RPi Imager only on that? :shrug:   (I know that's not what you're asking for, but maybe it's an easy workaround?)

danorum, 2022-03-26:
I always get nervous and triple check the drive. I would like a way to add a file to the root of the drive that would exclude it from the list. Maybe /noimager or something like that.

lurch, 2022-03-29:
The other "easy workaround" would be to unplug your 12 TB USB drive before running RPi Imager ? :shrug:

robbieb43, 2022-03-29:
Whilst the suggestions are appreciated, I am afraid they miss the point. There are any number of ways you can avoid this occurring with thought and planning but that is not why the feature was requested.  I may be unique here, but at times I act without thought and planning - perhaps when I have had an SD card fail and I am under pressure to restore my Pi’s function whilst trying to do what ever it was before it failed. A simple means to avoid this easily made mistake is what I had hoped for.

cillian64, 2023-10-19:
I don't think there's a good way around this, this is always going to be a dangerous tool and beyond excluding internal drives there's no good way to work out what a user does or doesn't want to overwrite.  If we add something like a blocklist then that encourages the user to not check carefully and makes it even more likely that they will overwrite the wrong drive if something changes or the blocklist doesn't work.  This might be a useful feature for a couple of people but I don't think it's a common enough case to justify the extra complexity.

closed 2023-10-19 by cillian64