raspberrypi/rpi-imager #294: imager not opening

ImLorio, 2021-10-18:
Instant crash when I launch `rpi-imager.exe`
```
Log Name:      Application
Source:        Application Error
Date:          10/18/2021 9:02:17 PM
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      lorioDesktop
Description:
Faulting application name: rpi-imager.exe, version: 1.6.2.0, time stamp: 0x00000000
Faulting module name: nvwgf2um.dll, version: 30.0.14.9613, time stamp: 0x615df3ad
Exception code: 0xc0000005
Fault offset: 0x00413f62
Faulting process id: 0x7f50
Faulting application start time: 0x01d7c452ab4254f7
Faulting application path: C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe
Faulting module path: C:\WINDOWS\System32\DriverStore\FileRepository\nv_dispi.inf_amd64_5292bbfbf575e2d2\nvwgf2um.dll
Report Id: 78bb7fc0-acfc-44cf-adc2-a502c9ca5196
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2021-10-18T19:02:17.5242225Z" />
    <EventRecordID>172365</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>lorioDesktop</Computer>
    <Security />
  </System>
  <EventData>
    <Data>rpi-imager.exe</Data>
    <Data>1.6.2.0</Data>
    <Data>00000000</Data>
    <Data>nvwgf2um.dll</Data>
    <Data>30.0.14.9613</Data>
    <Data>615df3ad</Data>
    <Data>c0000005</Data>
    <Data>00413f62</Data>
    <Data>7f50</Data>
    <Data>01d7c452ab4254f7</Data>
    <Data>C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe</Data>
    <Data>C:\WINDOWS\System32\DriverStore\FileRepository\nv_dispi.inf_amd64_5292bbfbf575e2d2\nvwgf2um.dll</Data>
    <Data>78bb7fc0-acfc-44cf-adc2-a502c9ca5196</Data>
  </EventData>
</Event>
```

lurch, 2021-10-18:
From a quick google-search, `nvwgf2um.dll` seems to be part of the NVIDIA driver stack. Are your graphics-card drivers up to date?

tdewey-rpi, 2024-02-06:
Closing as stale, no update since '21.

If you see this with v1.8.5, please raise a new issue.

closed 2024-02-06 by tdewey-rpi