raspberrypi/rpi-imager #298: "Error creating firstrun.sh on FAT partition" on Ubuntu 20.04 LTS installed from snap (v1.6.2)

CaptainJack42, 2021-11-03:
I'm trying to create a bootable SD Card with raspbian Lite on my Laptop with Ubuntu 20.04 LTS. I downloaded the latest version (v1.6.2) from snap (i tried channel stable and edge, bot having the same issue). Flashing the SD looks fine, but it'll say "Error creating firstrun.sh on FAT partition" after finishing the verification and indeed the boot partition doesn't have a `firstrun.sh`file. 

The imager works perfectly when installed from the .deb package (downloaded from the official website, version 1.6.2) though. So my guess is that the snap package has some unmet dependencies which aren't installed automatically. The .deb package also had some which i "manually" installed (with `apt --fix-broken install`) :

```sh
The following additional packages will be installed:
  libqt5quickcontrols2-5 libqt5quicktemplates2-5 qml-module-qtgraphicaleffects
  qml-module-qtquick-controls2 qml-module-qtquick-layouts
  qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtquick2
The following NEW packages will be installed:
  libqt5quickcontrols2-5 libqt5quicktemplates2-5 qml-module-qtgraphicaleffects
  qml-module-qtquick-controls2 qml-module-qtquick-layouts
  qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtquick2
```

lurch, 2021-11-04:
> The imager works perfectly when installed from the .deb package (downloaded from the official website, version 1.6.2) though. So my guess is that the snap package has some unmet dependencies which aren't installed automatically.

You'll need to report that at https://github.com/popey/imager-snap/issues

CaptainJack42, 2021-11-04:
> You'll need to report that at https://github.com/popey/imager-snap/issues

oh sry, didn't see there was a different repo for the snap package. Thx for the reply though, gonna report it there.

closed 2021-11-04 by CaptainJack42