raspberrypi/rpi-imager #299: [Feature Request] Support for 7-Zip archives

MichaIng, 2021-11-07:
With 7-Zip v21, official Linux CLI binaries are available: https://www.7-zip.org/

Debian already maintains packages for Bookworm: https://packages.debian.org/bookworm/7zip

We hence expect the official 7-Zip to become more popular on Linux platforms, replacing the outdated `p7zip` 3rd party port and being a notable extension or alternative to the `xz` format in certain use cases.

And of course I have a personal interest as [DietPi](https://github.com/MichaIng/DietPi) ships images as 7-Zip archives.

7-Zip is generally supported by [libarchive](https://github.com/libarchive/libarchive#supported-formats), the library used by RPi-Imager. Also fragments for 7-Zip support are present in the RPi-Imager code already. But 7z files cannot currently be selected as custom image.

I tried to add it, but reading the archive fails with this error: https://github.com/libarchive/libarchive/blob/ec5b86b/libarchive/archive_read.c#L238-L242
I guess something would need to be changed in the local file extractor, but I wasn't able detect the exact issue yet. Generally the single file vs multi file archive handling that is used for `zip` archives makes sense for `7z` as well, as it can contain multiple files, compared `gz`/`bz2`/`xz`: https://github.com/raspberrypi/rpi-imager/blob/qml/imagewriter.cpp#L188-L195

tdewey-rpi, 2024-02-06:
Rejecting feature request.

We don't see wide demand for 7z support, and this issue has seen no additional comments or reactions since '21. I'm happy to review a PR adding it, but won't progress this otherwise.

closed 2024-02-06 by tdewey-rpi

MichaIng, 2024-02-06:
In the meantime we also moved to `xz`-compressed images, so no demand from our end anymore either.