raspberrypi/rpi-imager #300: Saving cached download for use later.

ccheath, 2021-11-08:
Is it possible to save the cached download for use later? or for use with another imaging application?
I have found the location of the file (lastdownload.cache) but I'm unsure of its true format/extension.

DavidGriffith, 2022-03-28:
On top of this, it would be nice to be able to specify a location where downloads will be kept.  Maybe it would be nice to store them on a different machine on the local network.

shodanx2, 2022-05-02:
This is quite incredible 
lastdownload.cache

and it gets overwritten every time you change OS 

These are 1gb and more file and you're just redownloading them again ?

You people didn't live with 28kbps modems and it shows !  We'd never just overwrite files willy nilly like that !
Anyway, real simple fix, don't call it lastdownload.cache
Call it Raspberry Pi OS (32-bit) debian bullseye 2022-04-04.img

How has that not been changed since Nov 2001 ?!

Guess most people don't have bad internet like me !

maxnet, 2022-05-02:
>You people didn't live with 28kbps modems and it shows !

Having worked at a dial-up ISP as a teenager, I do remember those.
Time changes though...

I am certain that if we would cache multiple files we would have others complaining that they cannot miss the disk space though.
Keep in mind that users with say an entry-level Mac have almost no disk space to spare.

shodanx2, 2022-05-02:
Ok, then also add this in the persistent settings 

[image]

As long as the default stops being to delete multi gigabit sized downloads everytime you change the target OS !

lurch, 2022-05-03:
> How has that not been changed since Nov 2001 ?!

Huh? The Raspberry Pi itself only launched in Feb 2012.

I think if I had a "bad" internet connection, I'd download the image files I wanted manually (e.g. from https://www.raspberrypi.com/software/operating-systems/ ) and then select the (already-downloaded) image I want with the "Use custom" option in RPi Imager?

shodanx2, 2022-05-03:
@lurch 

That was supposed to read Nov 2021, start date of this current issue.

I don't think the solution to "if you have poor internet", should be "then use other software".

I get the feeling programmers that live in cities with gigabit internet, don't realize how annoying it is when software can't cope with less than perfect internet.

lurch, 2022-05-03:
> I get the feeling programmers that live in cities with gigabit internet, don't realize how annoying it is when software can't cope with less than perfect internet.

I remember **many** years (long before the days of BitTorrent) when I had dial-up internet, which would often cut out in the middle of a large download. Even back then I was using a dedicated "download manager" program that was able to queue downloads (so that no more than two things were being downloaded at once), and resume interrupted downloads (from where they left off) after the modem reconnected. I certainly wouldn't expect software as general as RPi Imager to offer features like that? :shrug: 

But anyway, that's getting a bit OffTopic!

shodanx2, 2022-05-03:
In this case that would be like the download manager deleting the first file when the second one started downloading.

As someone who downloaded pirated quake 2 on 28.8kbps for a day and a half, that just feels wrong

yurenchen000, 2022-05-25:
nothing about just download (choose path and save image to local)` without write to tf card?
It's GB size, through network (Worry about download speed) & write to disk (worry about ssd write count).

D-314, 2023-08-27:
For windows (Win+R, Ctrl+V):
`%AppData%\..\Local\Raspberry Pi\Imager\cache`

ccheath, 2023-08-27:
Why can't we get an answer to what the format/extension of the download.cache file actually is?

maxnet, 2023-08-28:
>Why can't we get an answer to what the format/extension of the download.cache file actually is?

Because it is not fixed, and depends on what file you last downloaded.

E.g. RPI OS is currently provided as a .xz image.
But the LibreELEC folks serve their stuff as .gz instead.

Imager does not care about extension and can auto-detect .zst .xz .gz .zip or no compression at all.

ccheath, 2023-08-28:
Not sure why it took almost 2 years to get this answer, but thanks... I guess we can close this one now.

closed 2023-08-28 by ccheath

maxnet, 2023-08-29:
>Not sure why it took almost 2 years to get this answer

Do keep in mind that github issues are for bug reports, which this is not.
Questions are better asked on the forum.
Even though not everyone knows Imager's internals there, someone would have likely typed "file lastdownload.cache" to tell you the answer.
The "file" command is available on all Unix flavours including RPI OS and macOs and analyzes contents of files for magic.

shodanx2, 2023-08-29:
Is this issue fixed ? 
Doesn't RPI imager overwrite the downloaded image each time ?
I know in fast internet land that feels like nothing, but if you download an ISO for 6 hours on a monthly limited data plan, I think the default for RPI imager should be not to auto-delete the file every time you change.

It's an easy fix, just save the file as it's original file instead of naming them all the same cache filename.

Maybe that has been actually done ? I have not used RPi imager since Pis they have not been purchasable for years anyway.

D-314, 2023-08-29:
No, it's still overwriting the file. Moreover, even if you save it for future use, it will not work. Imager will request internet access and download the file again each time even if it already exists. lastdownload.cache only works if you flash the SD card several times with the same image without closing the Imager window.
So now it's really easier to download the *.img manually, but you won't have automatic Wi-Fi settings and SSH enabled.

maxnet, 2023-08-29:
>Moreover, even if you save it for future use, it will not work. Imager will request internet access and download the file again each time even if it already exists.

That's incorrect.
Imager do will re-use it, as long as the source file is still the same (has the same sha256 checksum).
(May not be the case if there is a new RPI OS version).

(If you are manually replacing lastdownload.cache with something else, make sure you update lastDownloadSHA256 in `~/.config/Raspberry Pi/Imager.conf` (or registry if using Windows) as well to match what you put there...
May be easier to just use the "use custom file" option in Imager...)