raspberrypi/rpi-imager #301: Advanced options causing error

bylov, 2021-11-08:
I'm using version 1.6.2 on Ubuntu 21.10 [installing Nextcloud](https://ubuntu.com/appliance/nextcloud/raspberry-pi) which works fine, but changing any of the advanced options, e.g. "Set hostname: raspberrypi.local" results in "Error creating firstrun.sh on FAT partition" when writing to a micro SD card (both via the internal card reader and also via a USB card reader).

[image]

The card is writable. In order to troubleshoot, I actually managed to copy an empty file called firstrun.sh to the card just when the drive was mounted after the writing process had finished, but then I got a new error about not being able to find "/config.txt".

lurch, 2021-11-08:
Are you using the snap package? If so, this might be a duplicate of #298 ?

bylov, 2021-11-09:
Yep, that worked. Thanks @lurch.

closed 2021-11-09 by bylov