raspberrypi/rpi-imager #302: Snap package fails with "cannot open storage device"

brunoais, 2021-11-10:
When I press the button "write". The snap package version fails with "cannot open storage device"

Nothing is logged to the console regarding this. This is what appears in the console:

```
Qt: Session management error: Could not open network socket
"Object path cannot be empty"
qt.qpa.xcb: QXcbConnection: XCB error: 11 (BadAlloc), sequence: 447, resource id: 301989893, major code: 149 (Unknown), minor code: 2
qt.qpa.xcb: QXcbConnection: XCB error: 9 (BadDrawable), sequence: 448, resource id: 301989903, major code: 149 (Unknown), minor code: 4
qt.qpa.xcb: QXcbConnection: XCB error: 4 (BadPixmap), sequence: 451, resource id: 301989903, major code: 54 (FreePixmap), minor code: 0
qt.qpa.xcb: QXcbConnection: XCB error: 136 (Unknown), sequence: 452, resource id: 301989904, major code: 134 (Unknown), minor code: 17
QObject::setParent: Cannot set parent, new parent is in a different thread
Available disk space for caching: 0 GB
Telemetry done. cURL status code = 0
```
[image]

lurch, 2021-11-11:
See https://github.com/raspberrypi/rpi-imager/issues/298#issuecomment-960565278

brunoais, 2021-11-11:
Thanks @lurch

Closed as duplicate of https://github.com/popey/imager-snap/issues/15

closed 2021-11-11 by brunoais

DAT4, 2023-02-26:
Some people said that you can resolve this by running rpi-imager as sudo, to get permissions, so that you can access the disk, located in dev. However this is not a very secure way to do it, and may lead to other issues in some window managers. 

Best solution is to add your user to `disk` group which is the group that has persmision to  disks.

```
usermod -aG disk your-username
```

after that, either logout/login or run `su your-username`  and then run rpi-imager from console

devkev, 2023-06-12:
Giving a regular unprivileged user account full raw disk access isn't great advice.

In my case, the problem/solution was https://github.com/raspberrypi/rpi-imager/issues/359#issuecomment-1248073096.