raspberrypi/rpi-imager #304: GNU/Linux: Allow seeing the name of the device when choosing the device to image

brunoais, 2021-11-11:
**Problem:**
Currently, for an advanced user who knows the device names, it's not straightforward to know which device to choose from the provided list. I end up attaching and removing the SD card a few times to make sure which one listed is the SD card.

**Suggestion:**
Add the path to the device being chosen to the devices list, so I can confirm the name of the device being written to.
I'm fine even if it requires flipping a configuration option or including a command line argument.

lurch, 2021-11-11:
Don't the device-names on Linux tend to change though, based on the order you plug in devices? Just speaking as a third-party, I find using the disk-size to be a much more reliable way of telling drives apart, on the occasions when I have multiple drives connected at once. :man_shrugging:

brunoais, 2021-11-11:
An SD card will always be an `mmcblkX` where X is a number.
An ATA drive will always be a `dx` where `x` is a letter
A SATA drive will always be an `sdx` where `x` is a letter
An NVMe drive will always be an `nvmex` where `x` is a number

Just by looking at the name, you can already distinguish between what's connected. Ofc if you have multiple SD cards, you may not know the number.
However, if you are trying to flash an SD card and you have other drives attached, the SD card will always be the `mmcblkX` drive, undoubtedly.

Makes sense?

lurch, 2021-11-11:
> the SD card will always be the `mmcblkX` drive

That only works for the built-in SD card slot on a Raspberry Pi, or for the built-in SD card slot on a laptop (that's internally connected via PCIe rather than via USB). In the majority of cases, I suspect that people are using RPi Imager with a USB SD card-reader; and on Linux any drive plugged in via USB (SD card-reader, Flash drive, HDD, SSD, or an internal-card-reader that's connected via USB) always appears as `sdx` (i.e. the same as SATA drives).

brunoais, 2021-11-12:
That's interesting. I went to try the USB dongle I have (I think that thing is already 10 years old) and it also registers the SD card as mmcblk (I now use the internal reader, though). Maybe it's something with newer dongles?

lurch, 2021-11-12:
That's also interesting. In all the years I've been using different USB drives and different USB card-readers on Linux, they've always shown up as `/dev/sdx` :man_shrugging: 
Anyway, sorry for dragging your original question so OffTopic! :laughing:

Managor, 2023-07-04:
I have a Thinkpad T410 with an integrated SD card reader. Whenever I plug a SD card in there, it registers as `mmcblk0` but `rpi-imager` is unable to see it. Either fix `rpi-imager` to be able to detect mmcblk devices or allow us to choose the device path ourselves.

lurch, 2023-07-15:
@Managor See #606

Managor, 2023-07-15:
Sort of a duplicate, but I'm happy this gets more attention

maxnet, 2023-07-17:
>Sort of a duplicate, but I'm happy this gets more attention

@Managor
Do you happen to run Arch Linux or a distribution using Arch packages like Manjoro?
As that is what the other two users reporting this issue have in common...

Managor, 2023-07-17:
Yeah, Arch. I doubt that's the cause though.

maxnet, 2023-07-17:
>Yeah, Arch. I doubt that's the cause though.

Do note that one of the other affected users mentioned in #610 that it did work a couple weeks ago...
Our software hasn't changed.
So I wouldn't rule out an Arch update to another package (like the Linux kernel, or lshw which we use to ask if the drive is removable, or systemd/udev) causing a change in behavior...

Care to try if the issue also appears on your hardware, if you run a live USB image of Fedora or Ubuntu?
(Can just "yum install rpi-imager" or "apt install rpi-imager" in the live environment, without installing things to disk).

Managor, 2023-07-19:
Alright. It was the issue. 

`lsblk -o name,rm,hotplug` shows `mmcblk0` as hotplugable on Kubuntu 22.04 whereas on Arch it does not.

MrSmoer, 2023-07-29:
There was a similar discussion about displaying the device path in #439, I for myself 'd find it most helpful if there was a timestamp (or something like "n seconds ago") when the device got plugged in. This way one wouldn't have to lsblk and it's helpful information for the average user.
I don't know how to keep going on this, because balena did not reply to my pr yet  balena-io-modules/drivelist#421

Managor, 2024-02-04:
I believe this might have been fixed on Arch's end. `mmcblk0` is now listed as hotplugable with `lsblk -o name,rm,hotplug`

tdewey-rpi, 2024-02-06:
Rejecting feature request.

As was pointed out earlier in this issue, the name of the created device cannot be relied upon to determine actual form factor for choosing the storage device.

The effective workaround was already captured by the OP - if you cannot distinguish based on the criteria available, hotplugging the device at the "Choose Storage" window should make it obvious.

closed 2024-02-06 by tdewey-rpi