raspberrypi/rpi-imager #306: Flash fails with download error message

pauloimon, 2021-11-13:
I'm trying to flash my Micro SD card with the Raspberry Pi OS Lite image, but I received the following error twice:

> Error Downloading: GnuTLS recv error (-9): Error decoding the received TLS packet. - Server IP: 46.235.231.145

What's wrong? It's a network issue?

--

Raspberry Pi Imager: v1.6.2
OS: Ubuntu 20.04.3 LTS

andygeorge, 2021-12-14:
I get the same thing. Try [this tip](https://github.com/raspberrypi/rpi-imager/issues/322#issuecomment-992580953), downloading the image directly (link is probably https://downloads.raspberrypi.org/raspios_lite_armhf/images/ for the Lite OS).

pauloimon, 2021-12-15:
Thanks for the answer, @andygeorge! I'll try this tip soon.

andygeorge, 2021-12-15:
> I do not get that error if I simply pull the plug, or drop all packets with iptables

I suspect it's an issue from the `downloads.raspberrypi.org` side, and is sending bad data/terminating the connection from that side, which would look different than a simple loss of network connectivity on your clientside.

lurch, 2021-12-15:
@pauloimon Was this a transient / temporary error, or is it still happening repeatedly? What type of internet-connection do you have?

maxnet, 2021-12-15:
>I suspect it's an issue from the downloads.raspberrypi.org side, and is sending bad data/terminating the connection from that 
>side, which would look different than a simple loss of network connectivity on your clientside.

If server was sending "bad data" I would also expect it to cause problems on other platforms than Ubuntu though.
(Although that GnuTLS is used instead of OpenSSL do can result in other error conditions/bugs)

And if server abruptly terminates the TCP connection without properly letting say TLS goodbye, you typically get different error message.
This is what it looks like if the remote webserver is killed while downloading:

[image]

pauloimon, 2021-12-15:
@maxnet

> You do have the latest gnutls version?

Yes, I have the same version installed here:

```
libgnutls30/focal-updates,focal-security,now 3.6.13-2ubuntu1.6 amd64 [instalado,automático]
```

> And you do are using the Imager .deb package from raspberrypi.org ?

I'm using the .deb version downloaded from [releases page](https://github.com/raspberrypi/rpi-imager/releases) on GitHub.

> On a normal Intel/AMD computer?

Yes, my machine has an AMD64 Intel CPU.

--

@lurch

> Was this a transient / temporary error, or is it still happening repeatedly?

I didn't try recently, but I can test it again.

pauloimon, 2021-12-17:
I tried to flash again and now everything worked normally!

[image]

I don't know what was happening, but I think the problem is solved now.

Thanks for the answers, guys!

closed 2022-03-27 by pauloimon